nerves-project/nerves #376: tar: Failed to set default locale

smahi, 2019-01-23:
### Environment

* Elixir version (`elixir -v`): 

`Elixir 1.7.4 (compiled with Erlang/OTP 21)`

* Nerves environment: (`mix nerves.env --info`)

`
|nerves_bootstrap| Environment Package List

  Pkg:         nerves_system_br
  Vsn:         1.6.5
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_system_rpi3
  Vsn:         1.6.1
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Docker, []}

  Pkg:         nerves_toolchain_arm_unknown_linux_gnueabihf
  Vsn:         1.1.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.5.0
  Type:        toolchain_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   rpi3
  MIX_ENV:      dev

NERVES_TOOLCHAIN is set to a path which does not exist:
/Users/smahi/.nerves/artifacts/nerves_toolchain_arm_unknown_linux_gnueabihf-darwin_x86_64-1.1.0

`

* Additional information about your host, target hardware or environment that
  may help

  * host = macOS high sierra 10.13.6
  * target = rpi3

### Current behavior

`
Resolving Nerves artifacts...
  Resolving nerves_system_rpi3
  => Trying /Users/smahi/.nerves/dl/nerves_system_rpi3-portable-1.6.1-8676273.tar.gz
** (MatchError) no match of right hand side value: {:error, "tar: Failed to set default locale\n"}
    lib/nerves/artifact/cache.ex:24: Nerves.Artifact.Cache.put/2
    lib/mix/tasks/nerves.artifact.get.ex:88: Mix.Tasks.Nerves.Artifact.Get.put_cache/2
    (elixir) lib/enum.ex:765: Enum."-each/2-lists^foreach/1-0-"/2
    (elixir) lib/enum.ex:765: Enum.each/2
    (mix) lib/mix/task.ex:316: Mix.Task.run_task/3
    (nerves_bootstrap) lib/mix/tasks/nerves/deps.get.ex:15: Mix.Tasks.Nerves.Deps.Get.run/1
    (mix) lib/mix/task.ex:316: Mix.Task.run_task/3
    (mix) lib/mix/task.ex:350: Mix.Task.run_alias/3

`

### Expected behavior

Properly build the firmware

[image]

By the way I've also set the locale : 

`$ export LC_ALL="ar.UTF-8"` 

and some time

`$ export LC_ALL="en_US.UTF-8"`

without success

fhunleth, 2019-01-23:
Ugh - I have no clue. What happens when you run `locale -a`? Do you see `en_US.UTF-8`? If not, I'm not sure how to add it on OSX, but hopefully it's possible to google.

I also feel like we should be using the `C` locale internally to avoid locale issues, but I know that caused one issue in the past that I'm forgetting the details on right now.

smahi, 2019-01-23:
@fhunleth ,  I fixed the issue using the following  command

`$ export LANG=en_US.UTF-8`
`$ export LC_ALL=$LANG`

Thank you for your help.

fhunleth, 2019-01-23:
Thanks for letting me know what fixed it.

closed 2019-01-23 by fhunleth

mahendersheri, 2020-12-07:
Thank you so much my error got fixed.