nerves-project/nerves #381: module Mix.Nerves.IO

laibulle, 2019-02-01:
### Environment

* Elixir version (`elixir -v`): 
```
Erlang/OTP 21 [erts-10.1] [source] [64-bit] [smp:8:8] [ds:8:8:10] [async-threads:1] [hipe]

Elixir 1.7.4 (compiled with Erlang/OTP 19)
```
* Nerves environment: (`mix nerves.env --info`): Get an error message

The behavior appear in Ubuntu 18.04 and debian jessie

Elixir is installed via ASDF.

### Current behavior

```bash
git clone https://github.com/nerves-project/nerves_system_rpi3
cd nerves_system_rpi3
mix deps.get
mix nerves.artifact
```

returns
```
== Compilation error in file lib/mix/tasks/nerves.info.ex ==
** (CompileError) lib/mix/tasks/nerves.info.ex:3: module Mix.Nerves.IO is not loaded and could not be found

could not compile dependency :nerves, "mix compile" failed. You can recompile this dependency with "mix deps.compile nerves", update it with "mix deps.update nerves" or clean it with "mix deps.clean nerves"
```

laibulle, 2019-02-01:
@mobileoverlord Thank you, it fixed my issue.

closed 2019-02-01 by laibulle

axelson, 2019-03-24:
This fixed this issue for me too, but I'm surprised because from the output of `mix archive.install hex nerves_bootstrap` it didn't look like it was making any changes (replacing 1.5.0 with 1.5.0):
```
jason@jdesktop /t/nerves_system_rpi3> mix archive.install hex nerves_bootstrap
Resolving Hex dependencies...
Dependency resolution completed:
New:
  nerves_bootstrap 1.5.0
* Getting nerves_bootstrap (Hex package)
All dependencies are up to date

08:12:33.535 [info]  Application nerves_bootstrap exited: :stopped
Compiling 11 files (.ex)
Generated nerves_bootstrap app
Generated archive "nerves_bootstrap-1.5.0.ez" with MIX_ENV=prod
Found existing entry: /home/jason/.asdf/installs/elixir/1.8.1-otp-21/.mix/archives/nerves_bootstrap-1.5.0
Are you sure you want to replace it with "nerves_bootstrap-1.5.0.ez"? [Yn]
* creating /home/jason/.asdf/installs/elixir/1.8.1-otp-21/.mix/archives/nerves_bootstrap-1.5.0
```

axelson, 2019-03-24:
Ah, nervermind, what had fixed it was I switched to my elixir 1.8.1 installation which had nerves_bootstrap already installed. (I thought I had switched away from it but I just moved my asdf `.tool-versions` file to the parent directory :man_facepalming:).

GregMefford, 2019-03-24:
Ah yeah, it’s initially pretty confusing how each version of Elixir has independent archives installed. Glad you found it!