nerves-project/nerves #384: Request: give descriptive error if artifact site not defined when building

nathanl, 2019-02-15:
This problem was my fault for not following directions, but nevertheless is a chance to add a message to make the process more user-friendly. :smile:

I've successfully customized a Nerves system to add `mpv`, following [the guide](https://github.com/nerves-project/nerves/blob/b591c4e3a49fc2ec9d2dc2626f2cd08381884d99/docs/Systems.md#customizing-your-own-nerves-system). It shows commenting out the `artifact_sites:` key under `defp nerves_package`.

When I tried to do the [Creating an Artifact](https://github.com/nerves-project/nerves/blob/b591c4e3a49fc2ec9d2dc2626f2cd08381884d99/docs/Systems.md#creating-an-artifact) step, I got an inscrutable error:

```
** (Mix) The Nerves Docker build_runner encountered an error while building:

-----

-----

See /Users/nathanl/code/experiments/lullabeam_rpi3/build.log.
```
That log file was empty.

Eventually I tried uncommenting the `artifact_sites`, and it fixed the problem.

I see now that "Creating an Artifact" says:

> ...you first need to define the artifact_sites before creating the artifact

so I shouldn't have run into this. But an error like "can't build artifact because `artifact_sites` is undefined" would be friendlier.

### Environment

* Elixir version (`elixir -v`):
Erlang/OTP 21 [erts-10.1] [source] [64-bit] [smp:8:8] [ds:8:8:10] [async-threads:1] [hipe]
Elixir 1.8.1 (compiled with Erlang/OTP 20)

* Nerves environment: (`mix nerves.env --info`)
|nerves_bootstrap| Environment Package List

  Pkg:         lullabeam_rpi3
  Vsn:         1.6.2
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Docker, []}

  Pkg:         nerves_system_br
  Vsn:         1.6.6
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_toolchain_arm_unknown_linux_gnueabihf
  Vsn:         1.1.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.5.0
  Type:        toolchain_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   target
  MIX_ENV:      dev

|nerves_bootstrap| Environment Variable List
  target:     target
  toolchain:  /Users/nathanl/.nerves/artifacts/nerves_toolchain_arm_unknown_linux_gnueabihf-darwin_x86_64-1.1.0
  system:     /Users/nathanl/.nerves/artifacts/lullabeam_rpi3-portable-1.6.2
  app:        /Users/nathanl/code/experiments/lullabeam_rpi3

|nerves_bootstrap| Loadpaths End
* Additional information about your host, target hardware or environment that
  may help

GregMefford, 2019-02-15:
Thanks for taking the time to report this! 🎈I think if you ran into this problem, there are others who also hit it and it would be great to add an error message like that.

This is where that error message is coming from, if you (or someone else reading this) have any interest in tracing it back up the call stack to find an appropriate place to do that kind of "pre-flight check" and add an error message.

https://github.com/nerves-project/nerves/blob/f1119301307d4db410d49a14cecdbc946f65801a/lib/nerves/artifact/build_runners/docker.ex#L216

GregMefford, 2019-03-20:
I tried a bunch of things (obviously including commenting out the `artifact_sites` config) and wasn't able to get it to fail. I'm thinking that maybe the error was somehow related to the Docker container/volume that we use to compile custom systems may have gotten out of sync on your machine and that was the source of the problem, but un-commenting these lines fixed it because it caused the checksum to change, which triggered a fresh artifact build.

I'm going to go ahead and close this for now, but if someone is able to get it to happen again, I'd be happy to help dig in some more and figure out what the root problem was.

closed 2019-03-20 by GregMefford