nerves-project/nerves #385: nerves-init-gadget by default?

ConnorRigby, 2019-02-20:
related to #350 It seems like there is a greater than zero level of interest in changing the behaviour of the new project generator from opting into nerves_init_gadget with `--init-gadget` to opting out with `--no-init-gadget`.

I don't have any particular strong opinions on this, but if we opt not too default it, i think it may need better advertisement so users can find it.

fhunleth, 2019-02-20:
I'm pretty much in agreement that defaulting to `--init-gadget` would lead to an easier first time experience.

axelson, 2019-02-20:
👍 from me too. Also it would be good to have at least a short description of what `nerves_init_gadget` is. Currently it just says:
```
Generate a project configured to use nerves_init_gadget

    mix nerves.new blinky --init-gadget
```

lawik, 2019-02-20:
This was sort of why I brought up me missing out on init gadget initially. I think a --no-init-gadget would probably be more user-friendly. It wouldn't educate users to realize that the init-gadget is a thing you can do without and that this is some nice stuff for a developer that might not strictly be something you feel like shipping.

So it sort of depends what the project wants to optimize towards. But I think optimizing for adoption and a nice onboarding experience is generally considered the way to go these days. I can imagine drawbacks but they aren't terribly concrete. Might be more clear to people who actually ship devices and nerves products.

stwf, 2019-02-20:
I agree! If people don't know what init-gadget is or does they should be using it!

mobileoverlord, 2019-02-25:
I just release [`nerves_bootstrap v1.5.0`](https://github.com/nerves-project/nerves_bootstrap/releases/tag/v1.5.0) which will include `nerves_init_gadget` by default.

closed 2019-02-25 by mobileoverlord