nerves-project/nerves #395: Customizing nerves system instructions appear to be out of date

axelson, 2019-03-21:
### Environment

* Elixir version (`elixir -v`): Elixir 1.8.1 (compiled with Erlang/OTP 21)
* Nerves environment: (`mix nerves.env --info`): 
```
|nerves_bootstrap| Environment Package List                                                                                  [0/56739]
  Pkg:         nerves_toolchain_ctng
  Vsn:         1.5.0
  Type:        toolchain_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_system_custom_rpi3
  Vsn:         1.7.0
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_arm_unknown_linux_gnueabihf
  Vsn:         1.1.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_system_br
  Vsn:         1.7.0
  Type:        system_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   custom_rpi3
  MIX_ENV:      dev

NERVES_SYSTEM is set to a path which does not exist:
/home/jason/.nerves/artifacts/nerves_system_custom_rpi3-portable-1.7.0

Try running `mix deps.get` to see if this resolves the issue by
downloading the missing artifact.
```
* Additional information about your host, target hardware or environment that
  may help

### Current behavior

After following the https://hexdocs.pm/nerves/systems.html#customizing-your-own-nerves-system instructions as well as I can I end up with an error after running `mix firmware`

```
jason@jdesktop ~/d/s/fw> mix firmware

Nerves environment
  MIX_TARGET:   custom_rpi3
  MIX_ENV:      dev

warning: module attribute @all_targets was set but never used
  /home/jason/dev/forks/nerves_system_rpi3/mix.exs:9

NERVES_SYSTEM is set to a path which does not exist:
/home/jason/.nerves/artifacts/nerves_system_custom_rpi3-portable-1.7.0

Try running `mix deps.get` to see if this resolves the issue by
downloading the missing artifact.
```

Also `mix deps.get` print out:
```
...

Resolving Nerves artifacts...
  Resolving nerves_system_custom_rpi3
  Skipping nerves_system_custom_rpi3
  Cached nerves_toolchain_arm_unknown_linux_gnueabihf
```
which may or may not be correct

### Expected behavior

`mix firmware` would compile the new custom system.

### Thoughts

I think that the instructions mentioning changing the `@all_targets` line in your system's `mix.exs` seem incorrect, because as far as I can tell `@all_targets` is usually part of your nerves project, not the nerves system.

Ideally an example up-to-date "custom" nerves system would exist that would show the patch that is needed from the `master` branch of `nerves_system_rpi3` (or any other system).

ConnorRigby, 2019-03-21:
this should be fixed by https://github.com/nerves-project/nerves/pull/394

axelson, 2019-03-21:
This is the patch that I'm currently using:

```
jason@jdesktop ~/d/f/nerves_system_rpi3> git show
commit 910e5a74048a60c9327c42b7514ab6a00e987798 (HEAD -> custom)
Author: Jason Axelson <jason.axelson@gmail.com>
Date:   Thu Mar 21 08:07:27 2019 -1000

    Customize

diff --git a/mix.exs b/mix.exs
index 1b150c2..5ce5cff 100644
--- a/mix.exs
+++ b/mix.exs
@@ -1,11 +1,14 @@
-defmodule NervesSystemRpi3.MixProject do
+defmodule CustomRpi3.MixProject do
   use Mix.Project

-  @app :nerves_system_rpi3
+  @app :nerves_system_custom_rpi3
   @version Path.join(__DIR__, "VERSION")
            |> File.read!()
            |> String.trim()

+  @all_targets [:rpi, :rpi0, :rpi2, :rpi3, :rpi3a, :bbb, :x86_64, :custom_rpi3]
+  @my_target_test 42
+
   def project do
     [
       app: @app,
@@ -34,9 +37,9 @@ defmodule NervesSystemRpi3.MixProject do
   defp nerves_package do
     [
       type: :system,
-      artifact_sites: [
-        {:github_releases, "nerves-project/#{@app}"}
-      ],
+      # artifact_sites: [
+      #   {:github_releases, "nerves-project/#{@app}"}
+      # ],
       build_runner_opts: build_runner_opts(),
       platform: Nerves.System.BR,
       platform_config: [
@@ -67,7 +70,7 @@ defmodule NervesSystemRpi3.MixProject do
       maintainers: ["Frank Hunleth", "Justin Schneck"],
       files: package_files(),
       licenses: ["Apache 2.0"],
-      links: %{"GitHub" => "https://github.com/nerves-project/#{@app}"}
+      links: %{"GitHub" => "https://github.com/axelson/#{@app}"}
     ]
   end
```

(The `@all_targets` and `@my_target` are incorrect but harmless because nothing uses them)

When I run `mix firmware` I get:
```
jason@jdesktop ~/d/s/fw> mix firmware

Nerves environment
  MIX_TARGET:   custom_rpi3
  MIX_ENV:      dev

warning: module attribute @my_target_test was set but never used
  /home/jason/dev/forks/nerves_system_rpi3/mix.exs:10

warning: module attribute @all_targets was set but never used
  /home/jason/dev/forks/nerves_system_rpi3/mix.exs:9

NERVES_SYSTEM is set to a path which does not exist:
/home/jason/.nerves/artifacts/nerves_system_custom_rpi3-portable-1.7.0

Try running `mix deps.get` to see if this resolves the issue by
downloading the missing artifact.
```

<details>
  <summary>`mix deps.get` prints:</summary>
```
jason@jdesktop ~/d/s/fw> mix deps.get
warning: module attribute @my_target_test was set but never used
  /home/jason/dev/forks/nerves_system_rpi3/mix.exs:10

warning: module attribute @all_targets was set but never used
  /home/jason/dev/forks/nerves_system_rpi3/mix.exs:9

Resolving Hex dependencies...
Dependency resolution completed:
Unchanged:
  artificery 0.4.1
  bunt 0.2.0
  certifi 2.5.1
  combine 0.10.0
  credo 1.0.3
  crontab 1.1.5
  dialyxir 1.0.0-rc.4
  distillery 2.0.12
  dns 2.1.2
  elixir_make 0.5.2
  erlex 0.2.1
  gettext 0.16.1
  hackney 1.15.1
  idna 6.0.0
  input_event 0.4.0
  jason 1.1.2
  mdns 1.0.2
  metrics 1.0.1
  mimerl 1.2.0
  nerves 1.4.1
  nerves_firmware_ssh 0.4.2
  nerves_init_gadget 0.6.0
  nerves_network 0.5.5
  nerves_network_interface 0.4.6
  nerves_runtime 0.9.3
  nerves_system_br 1.7.0
  nerves_system_linter 0.3.0
  nerves_toolchain_arm_unknown_linux_gnueabihf 1.1.0
  nerves_toolchain_ctng 1.5.0
  nerves_wpa_supplicant 0.5.2
  one_dhcpd 0.2.1
  parse_trans 3.3.0
  ring_logger 0.6.1
  scenic 0.9.0
  scenic_driver_nerves_rpi 0.9.3
  scenic_driver_nerves_touch 0.9.1
  sched_ex 1.0.4
  shoehorn 0.4.0
  socket 0.3.13
  ssl_verify_fun 1.1.4
  system_registry 0.8.2
  timex 3.5.0
  toolshed 0.2.6
  tzdata 0.5.19
  uboot_env 0.1.0
  unicode_util_compat 0.4.1

Nerves environment
  MIX_TARGET:   custom_rpi3
  MIX_ENV:      dev

Resolving Nerves artifacts...
  Resolving nerves_system_custom_rpi3
  Skipping nerves_system_custom_rpi3
  Cached nerves_toolchain_arm_unknown_linux_gnueabihf
```
</details>

ConnorRigby, 2019-03-21:
Can i see your `deps` from your project's mix.exs?

ConnorRigby, 2019-03-21:
```
nerves_system_br 1.7.0
nerves_system_linter 0.3.0
nerves_toolchain_arm_unknown_linux_gnueabihf 1.1.0
nerves_toolchain_ctng 1.5.0
```

I feel as if i should see `nerves_system_custom_rpi3 1.7.1` in here.

axelson, 2019-03-21:
> Can i see your deps from your project's mix.exs?

https://github.com/axelson/scenic_asteroids/tree/upgrade-nerves
(look in the fw/mix.exs file)

and here's the system I'm using:
https://github.com/axelson/nerves_system_rpi3/tree/broken-custom-system

(Note I could have messed something up when following the instructions, but I think I followed correctly)

axelson, 2019-03-22:
> I feel as if i should see `nerves_system_custom_rpi3 1.7.1` in here.

Here's the `deps` directly:
```
  defp deps do
    [
      {:nerves, "~> 1.3", runtime: false, targets: @all_targets},
      {:nerves_firmware_ssh, ">= 0.0.0", targets: @all_targets},
      {:shoehorn, "~> 0.4"},
      {:ring_logger, "~> 0.4"},
      {:toolshed, "~> 0.2"},

      {:dialyxir, "1.0.0-rc.4", only: :dev, runtime: false},
      {:credo, ">= 0.0.0", only: [:dev, :test], runtime: false},

      {:play, path: "../play"},

      {:nerves_runtime, "~> 0.6", targets: @all_targets},
      {:nerves_init_gadget, "~> 0.4", targets: @all_targets},
      {:scenic_driver_nerves_rpi, "~> 0.9", targets: @all_targets},
      {:scenic_driver_nerves_touch, "~> 0.9", targets: @all_targets},
      # {:nerves_system_rpi3, "~> 1.0", runtime: false, targets: :rpi3}
      {:nerves_system_custom_rpi3, path: "~/dev/forks/nerves_system_rpi3", runtime: false, targets: :custom_rpi3}
    ]
  end
```

You don't see `nerves_system_custom_rpi3` in there directly because it is a path dependency.

axelson, 2019-03-22:
Okay, so to aid in debugging I'm now using a local path-based version of nerves (although this probably doesn't matter). Also I'm running `mix compile` between `mix deps.get` and `mix firmware` (I would've assumed that `mix firmware` would run the equivalent of `mix compile`). So running `mix compile` ran some of the associated make files which showed that I was missing `cpio` and `wget`, so I've installed those and building the system now appears to be progressing.

GregMefford, 2019-03-22:
My guess is that `mix deps.get` isn’t properly trying to download/build the system because it’s a path dependency. I normally go to the system directory and `mix nerves.artifact` to make sure it builds successfully and exists in the artifacts directory. I think the source of the error about the `NERVES_SYSTEM` directory not existing is due to the system compilation having been skipped. I agree that this isn’t expected behavior and we should fix it.

GregMefford, 2019-03-22:
I checked out your branches of your project and custom system (using the branches you mentioned) and was able to get it to build firmware, first using the official `rpi3` system and then switching to your custom one. I was "hoping" that there would be some corruption of the environment when switching from the official system to your modified one, since the directory names are the same, but it just worked for me. 😞 

Have you tried `rm -rf _build deps` from your `scenic_asteroids/fw` project directory? You can also try `mix nerves.clean --all` to clear the cache of Nerves artifacts and start fresh. I think something somewhere is getting mixed up in your build environment, but it's not clear _where_.

axelson, 2019-03-22:
Hmmm, running into an error when I try `mix nerves.clean --all`
```
jason@jdesktop ~/d/s/fw> mix nerves.clean --all
Cleaning Nerves Package nerves_toolchain_arm_unknown_linux_gnueabihf
** (UndefinedFunctionError) function Nerves.Toolchain.CTNG.clean/1 is undefined (module Nerves.Toolchain.CTNG is not available)
    Nerves.Toolchain.CTNG.clean(%Nerves.Package{app: :nerves_toolchain_arm_unknown_linux_gnueabihf, build_runner: {Nerves.Artifact.BuildRunners.Local, []}, compilers: [:nerves_package, :yecc, :leex, :erlang, :elixir, :xref, :app], config: [platform_config: [defconfig: "defconfig"], target_tuple: :arm_unknown_linux_gnueabihf, artifact_sites: [github_releases: "nerves-project/toolchains"], checksum: ["mingw32_x86_64_defconfig", "defconfig", "README.md", "LICENSE", "mix.exs", "VERSION"]], dep: :hex, dep_opts: [], path: "/home/jason/dev/scenic_asteroids/fw/deps/nerves_toolchain_arm_unknown_linux_gnueabihf", platform: Nerves.Toolchain.CTNG, type: :toolchain, version: "1.1.0"})
    (elixir) lib/enum.ex:769: Enum."-each/2-lists^foreach/1-0-"/2
    (elixir) lib/enum.ex:769: Enum.each/2
    (mix) lib/mix/task.ex:331: Mix.Task.run_task/3
    (mix) lib/mix/cli.ex:79: Mix.CLI.run_task/2
```

So far I've been unable to reproduce the error when using a local nerves install. And once I get it to build once it works until I blow away the `~/.nerves` directory completely.

When it works `mix.deps` says:
```
Resolving Nerves artifacts...
  Cached nerves_system_custom_rpi3
  Cached nerves_toolchain_arm_unknown_linux_gnueabihf
```

But when it doesn't is when it's printing:
```
Resolving Nerves artifacts...
  Resolving nerves_system_custom_rpi3
  Skipping nerves_system_custom_rpi3
  Cached nerves_toolchain_arm_unknown_linux_gnueabihf
```

I have been doing a `rm -rf _build deps` in the `scenic_asteroids/fw` project each time I test.

GregMefford, 2019-03-22:
Yeah that `skipping` message seems likely to be related to the problem.

GregMefford, 2019-03-22:
Ah, I can see a code path where commenting out `artifact_sites` in your custom system might cause this behavior. Can you try uncommenting that and check, to see if it helps? Ideally, point it to your own GitHub repo instead of the official one, so it doesn’t get confused.

axelson, 2019-03-23:
Well with uncommenting it the `artifact_sites` I get:
```
==> fw
Resolving Nerves artifacts...
  Resolving nerves_system_custom_rpi3
  => Trying https://github.com/axelson/nerves_system_rpi3/releases/download/v1.7.0/nerves_system_custom_rpi3-portable-1.7.0-798C3FD.tar.gz
     Status 404 Not Found
  => Trying https://github.com/axelson/nerves_system_rpi3/releases/download/v1.7.0/nerves_system_custom_rpi3-portable-1.7.0-798C3FDB91A700EC8D100204475E4F95F0E9E7B560649778AD5BAF4CE69B466A.tar.gz
     Status 404 Not Found
  => no_result
  Cached nerves_toolchain_arm_unknown_linux_gnueabihf
```

Which makes sense since I haven't actually created any releases on GitHub (and I'd prefer not to because I'm just testing a small change).

And then `mix firmware` still unable to find my custom system:
```

axelson, 2019-03-23:
> I think the source of the error about the `NERVES_SYSTEM` directory not existing is due to the system compilation having been skipped. I agree that this isn’t expected behavior and we should fix it.

Where is the system compilation supposed to be triggered? Is that during `mix deps.get`?

GregMefford, 2019-03-28:
Ok, I've been able to confirm that it's not falling-back to a local build when the artifact file doesn't exist. When the system artifacts isn't present, `mix firmware` gives the error message mentioned above.

When running `mix deps.get` with the `artifact_sites` option commented-out altogether, you get this output (**TL;DR: it skips trying to cache because there's no `artifact_sites` defined**):

```
warning: module attribute @my_target_test was set but never used
  /Users/gmefford/projects/nerves/nerves_issue_395/nerves_system_rpi3/mix.exs:10

warning: module attribute @all_targets was set but never used
  /Users/gmefford/projects/nerves/nerves_issue_395/nerves_system_rpi3/mix.exs:9

Resolving Hex dependencies...
Dependency resolution completed:
Unchanged:
  -->8-- Snip -->8--

Nerves environment
  MIX_TARGET:   custom_rpi3
  MIX_ENV:      dev

Resolving Nerves artifacts...
  Resolving nerves_system_custom_rpi3
  Skipping nerves_system_custom_rpi3
  Cached nerves_toolchain_arm_unknown_linux_gnueabihf
```

When you do have `artifact_sites` defined, but there's not an actual artifact for the version/hash requested, you get this when you `mix deps.get` (**TL;DR: it 404s, but doesn't try to build locally**):

```
warning: module attribute @my_target_test was set but never used
  /Users/gmefford/projects/nerves/nerves_issue_395/nerves_system_rpi3/mix.exs:10

warning: module attribute @all_targets was set but never used
  /Users/gmefford/projects/nerves/nerves_issue_395/nerves_system_rpi3/mix.exs:9

Resolving Hex dependencies...
Dependency resolution completed:
Unchanged:
  -->8-- Snip -->8--

Nerves environment
  MIX_TARGET:   custom_rpi3
  MIX_ENV:      dev

Resolving Nerves artifacts...
  Resolving nerves_system_custom_rpi3
  => Trying https://github.com/nerves-project/nerves_system_custom_rpi3/releases/download/v1.7.0/nerves_system_custom_rpi3-portable-1.7.0-A37B308.tar.gz
     Status 404 Not Found
  => Trying https://github.com/nerves-project/nerves_system_custom_rpi3/releases/download/v1.7.0/nerves_system_custom_rpi3-portable-1.7.0-A37B30843FF9245568F3339D0A3BE0EBA232D48EA82AC2918EFE33608C17A9F6.tar.gz
     Status 404 Not Found
  => no_result
 ```

I believe the easy fix for this is to add `nerves: [compile: true]` to the dependency line for any custom systems that you want to build locally if they fail to fetch an artifact. @ConnorRigby is working on a PR to improve the error messages/docs to make this more discoverable (I'd seen it before but forgot what the option was and couldn't find it in the docs either).

heeton, 2019-12-18:
@GregMefford I'm hitting the same issue, have I understood your suggested fix correctly? 

My firmware dependency line looks like this: 
```
{:python_rpi0,
       nerves: [compile: true], path: "../../python_rpi0", runtime: false, targets: :python_rpi0}
``` 

But I'm still hitting the same problems as above. Running `mix deps.get` shows
```
Nerves environment
  MIX_TARGET:   python_rpi0
  MIX_ENV:      dev

Resolving Nerves artifacts...
  Resolving python_rpi0
  => Trying https://github.com/heeton/nerves_system_rpi0/releases/download/v1.10.0/python_rpi0-portable-1.10.0-3E0BBB5.tar.gz
     Status 404 Not Found
  => Trying https://github.com/heeton/nerves_system_rpi0/releases/download/v1.10.0/python_rpi0-portable-1.10.0-3E0BBB5EBE94954807852A4EE1E6A9DFE31A439954F0266B535E842D43ED91C2.tar.gz
     Status 404 Not Found
  => no_result
  Cached nerves_toolchain_armv6_rpi_linux_gnueabi
```

And then `mix firmware` shows
```
Nerves environment
  MIX_TARGET:   python_rpi0
  MIX_ENV:      dev

NERVES_SYSTEM is set to a path which does not exist:
/Users/alex/.nerves/Users/alex/Work/Code/Elixir/Nerves/appleyard/appleyard_firmware/deps/nerves/artifacts/python_rpi0-portable-1.10.0

Try running `mix deps.get` to see if this resolves the issue by
downloading the missing artifact.
```

GregMefford, 2019-12-19:
Hmm yes, it looks like that’s correct. I’m suspicious though that there is something still interfering somehow on your machine and making it not work. It’s possible that there is a cache of the official `nerves_system_rpi0`. Make sure in your system’s `mix.exs`, that you have renamed it to something unique on the line near the top where it defines `@app`.

You could also try `mix nerves.clean nerves_system_rpi0` to clear out any local caches for that system.

hoblin, 2020-03-20:
Hi, I have the same issue with the customization guide. @alexslade did you found the way to fix that 404 error?

jjcarstens, 2020-04-01:
**TL;DR** - So it looks like the problem here is order of operations when running `mix firmware` which doesn't allow the `nerves: [compile: true]` path to run when it hasn't been compiled. The work around is to keep the `nerves: [compile: true]` and run `compile` first (i.e. `mix do compile, firmware`)

The whole path is:
1. [`nerves.loadpaths` task is prepended to run first](https://github.com/nerves-project/nerves_bootstrap/blob/master/lib/nerves_bootstrap/aliases.ex#L45) before everything else, including compiling deps
2. [the `nerves.loadpaths` task then calls the `Nerves.Env.bootstrap()](https://github.com/nerves-project/nerves_bootstrap/blob/master/lib/mix/tasks/nerves/loadpaths.ex#L16)
3. ['Nerves.Env.bootstrap()` fails with the error above](https://github.com/nerves-project/nerves/blob/master/lib/nerves/env.ex#L358) because the nerves env variables aren't set, because the system isn't compiled, because deps compilation doesn't come until later

A good old fashioned catch-22.

The fix will probably be to check the nerves package configuration if it is a `:path` type and not `:hex, then check `nerves: [compile: true]` option, and if all passes, somehow mark that it will be compiled later and do the checks again. This will prob take a bit for me to decide how to implement.

In the meantime, if you keep the `nerves: [compile: true]` option and run `compile` first (like `mix do deps.get, compile, firmware`) then it _should_ work

fhunleth, 2020-10-03:
Looks like this should have been closed with #506.

closed 2020-10-03 by fhunleth