nerves-project/nerves #396: init terminating in do_boot (cannot expand $ERTS_LIB_DIR in bootfile)

ychaudhari7, 2019-03-22:
### Environment

* Elixir version (`Elixir 1.8.1 (compiled with Erlang/OTP 21)`):
* Nerves environment: (`|nerves_bootstrap| Environment Package List

  Pkg:         nerves_system_br
  Vsn:         1.7.0
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_system_rpi0
  Vsn:         1.7.0
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_armv6_rpi_linux_gnueabi
  Vsn:         1.1.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.5.0
  Type:        toolchain_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   rpi0
  MIX_ENV:      dev

|nerves_bootstrap| Environment Variable List
  target:     rpi0
  toolchain:  /home/ganesh/.nerves/artifacts/nerves_toolchain_armv6_rpi_linux_gnueabi-linux_x86_64-1.1.0
  system:     /home/ganesh/.nerves/artifacts/nerves_system_rpi0-portable-1.7.0
  app:        /home/ganesh/exercism/myelixircode/nerves_examples/hello_network

|nerves_bootstrap| Loadpaths End
`)
* Additional information about your host, target hardware or environment that
  may help

### Current behavior

> init terminating in do_boot (cannot expand $ERTS_LIB_DIR in bootfile)

I did `mix nerves.release.init` but cannot build firmware I get 
>** (Mix) Nerves encountered an error. %IO.Stream{device: :standard_io, line_or_bytes: :line, >raw: true}
 
### Expected behavior
I cloned the neves_network it should boot and connect to network on `rpi0`

mobileoverlord, 2019-03-22:
It looks like you are having trouble running the `hello_network` example from [nerves_examples](https://github.com/nerves-project/nerves_examples/tree/master/hello_network). I just cloned this and ran it on a rpi0. I don't get the same message you are getting. Are you getting `Nerves encountered an error.` when you call `mix firmware`? If so, is there any output prior to that?

GregMefford, 2019-03-22:
I'm not exactly sure what the problem is, but sometimes if you're switching Erlang/Elixir versions, things can get out-of-sync with your project. Something to try that might help: clear out the `_build` and `deps` directories and try building fresh again, like this:

```bash
# From your project directory
rm -rf _build deps
mix deps.get
mix firmware
```

ychaudhari7, 2019-03-23:
@mobileoverlord I did on fresh project and it worked and I did not run `mix nerves.release.init` not needed because it was already generated in *hello_network* is it  a good practice to generate `mix nerves.release.init` always instead of default generator.

ychaudhari7, 2019-03-23:
@GregMefford It worked somehow once when I deleted while `deps` directory and  it did it again then same issue I think what I am doing wrong is calling `mix deps.get` forgetting setting `MIX_TARGET` then I set and run  `mix deps.get` again. Will it mess up with build without clearing first fetched by `MIX_TARGET=HOST`?

closed 2019-03-28 by ychaudhari7