nerves-project/nerves #402: Docker build_runner permissions error

brandonmenc, 2019-03-29:
### Environment

* Elixir version: `Elixir 1.8.0 (compiled with Erlang/OTP 20)`

* Nerves environment:

```
|nerves_bootstrap| Environment Package List

  Pkg:         custom_rpi3
  Vsn:         1.7.0
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Docker, []}

  Pkg:         nerves_system_br
  Vsn:         1.7.0
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_toolchain_arm_unknown_linux_gnueabihf
  Vsn:         1.1.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.5.0
  Type:        toolchain_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   target
  MIX_ENV:      dev
```

### Current behavior

`mix compile` or `mix nerves.artifact` produces the following error:

```
|Elixir.Docker| Starting Build... (this may take a while)

** (Mix) The Nerves Docker build_runner encountered an error while building:

-----
Setting User
UID: 501
GID: 20
Switching user
/nerves/env/platform/create-build.sh: line 81: /nerves/env/custom_rpi3/Config.in: Permission denied

-----
```

### Expected behavior

A custom system being built.

mobileoverlord, 2019-03-29:
It looks like this is an issue with docker trying to write the Config.in file back to the host and it doesn't have permission. What host are you on, Mac OS or Linux?

brandonmenc, 2019-03-29:
macOS

mobileoverlord, 2019-03-29:
I was able to run through a clean build on macOS without getting this error. What version of docker are you running? I did notice that my docker disk was out of space when I ran it so I had to stop docker, toss the disk, and start docker again. This forced the system to pull the image again. Not sure if you could give this a go. The nice thing about docker is that it's not too tough to rebuild the disk.

brandonmenc, 2019-03-29:
Docker version 18.09.2, build 6247962

Deleted my docker disk and tried again. Got the same error.

brandonmenc, 2019-04-02:
Problem solved. Operator error. Problem was similar to this one: https://github.com/moby/moby/issues/22114

closed 2019-04-02 by brandonmenc