nerves-project/nerves #404: Update documentation for overwriting files in boot partition

deanchouinard, 2019-04-02:
Recently, it was mentioned on Slack that the documentation for overwriting files in the boot partition needs to be updated. I took a look and propose these [changes](https://github.com/deanchouinard/nerves/commit/244d2876deb18db00d759374f683464c01428086).

ConnorRigby, 2019-04-02:
Those changes look good to me. Feel free to open a PR and we can get them reviewed and merged

fvasquez, 2019-07-24:
Dean's changes appear to have already been merged to master.

```
$ git log -- Advanced\ Configuration.md
commit 5ee1e1a51a5e05afd465aec7228e3c944c1e113f
Author: Dean Chouinard <deanchouinard@gmail.com>
Date:   Sun Mar 31 16:09:56 2019 -0400

    Adds example of overwriting files in boot partition
```

Someone (maybe @ConnorRigby) should close this issue.

fvasquez, 2019-07-24:
I see now that most of the open issues in this repo have already been resolved.  I'm guessing the maintainers of this project don't relay on GitHub issues for bug tracking.  Sorry to bother you all.  Keep up the good work.

mobileoverlord, 2019-07-24:
Thanks for the heads up 😁

closed 2019-07-24 by mobileoverlord