nerves-project/nerves #416: mix test produces config key warnings

fhunleth, 2019-05-02:
Warnings are emitted now for extraneous config keys. This triggers on `host` builds, so with a pretty basic `mix nerves.new` generated project, you'll get things like this:

```
You have configured application :nerves_runtime in your configuration file,
but the application is not available.

This usually means one of:

  1. You have not added the application as a dependency in a mix.exs file.

  2. You are configuring an application that does not really exist.

Please ensure :nerves_runtime exists or remove the configuration.

You have configured application :nerves_init_gadget in your configuration file,
but the application is not available.

This usually means one of:

  1. You have not added the application as a dependency in a mix.exs file.

  2. You are configuring an application that does not really exist.

Please ensure :nerves_init_gadget exists or remove the configuration.
```

GregMefford, 2019-06-08:
For what it's worth, I personally don't ever run in `MIX_TARGET=host` mode. If I want to test code on my development machine, I do so in a separate Mix project, and basically the Mix project that pulls in Nerves is only a deployment wrapper that I don't need to run except on the physical hardware. I'm not sure if that's what others are doing, but felt like I should mention it in case it guides any decisions about how commonly-used this feature is in the first place.

Do you have thoughts / opinions on how you think we should solve this? Here are a couple of options that come to mind for me:

* Change the project generator so that it splits some of these things out between a `host.exs` config file vs. one or more `target.exs` files. That way, we can move the `nerves_runtime` and `nerves_init_gadget` application configs into a non-`host` config and eliminate the warnings without changing anything else. This wouldn't automatically eliminate the warnings for people, but my guess is that people who see these warnings will know how to do this for themselves anyway. This change seems relatively easy to me.

* Change `nerves_runtime` and `nerves_init_gadget` so that they can safely run (in some limited way) in `host` mode. This may or may not be more useful for people who want to test against things that depend on these applications, assuming we had some kind of API compatibility while running in `host` mode. This sounds like more work than it's worth to me, but it would be nice to have if we ignore the effort involved.

mobileoverlord, 2020-03-23:
Starting with `nerves_bootstrap` ~> 1.6.0 the new project generator separates the config definitions into `config.exs` (all targets including host) and `target.exs` (all targets except host). The project config has also added `preferred_cli_target: [run: :host, test: :host]` to ensure that tests and runs are all done in the `host` target.

closed 2020-03-23 by mobileoverlord