nerves-project/nerves #428: Dhcpd with wlan mode 2

iancharters, 2019-06-18:
I have spent hours trying to get this to work configuring it in every possible combination I can imagine.  

Is it possible to create a host (mode 2) wlan with dhcpd?  I can create a wlan network from my rpi3 that I can connect to, but I cannot find the nerves device or ping nerves.local.

fhunleth, 2019-06-18:
What's your configuration look like? Is it something like this?

```
config :nerves_init_gadget,
  ifname: "wlan0",
  address_mode: :dhcp,
  mdns_domain: "nerves.local",
  node_name: node_name,
  node_host: :mdns_domain
```

iancharters, 2019-06-18:
I've managed to fix this.  I was specifying `:dhcpd` in `:nerves_init_gadget` but needed to do it again in `:nerves_network`.  This is my now working config just in case someone stumbles on this in the future.

```
config :nerves_init_gadget,
  ifname: "wlan0",
  address_method: :dhcpd,
  mdns_domain: "nerves.local",
  node_name: node_name,
  node_host: :mdns_domain

config :nerves_network,
  regulatory_domain: "CA"

config :nerves_network, :default,
  wlan0: [
    networks: [
      [
        ssid: "Pully",
        psk: "supersecret",
        key_mgmt: :"WPA-PSK",
        ipv4_address_method: :dhcpd,
        mode: 2
      ]
    ]
  ],
  eth0: [
    ipv4_address_method: :dhcp
  ]
```

I find the relationship between `:nerves_init_gadget` and `:nerves_network` a little confusing.  Are you guys open to people opening PRs for perhaps cleaning up docs or adding examples?

closed 2019-06-18 by iancharters