nerves-project/nerves #432: Failed to archive release (permission denied) when strip_debug_info=true

Manwholikespie, 2019-07-06:
### Environment

* Elixir version (`elixir -v`): `Elixir 1.8.1 (compiled with Erlang/OTP 21)`
* Nerves environment: (`mix nerves.env --info`)
```
|nerves_bootstrap| Environment Package List

  Pkg:         nerves_system_br
  Vsn:         1.7.2
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_system_rpi0
  Vsn:         1.7.2
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Docker, []}

  Pkg:         nerves_toolchain_armv6_rpi_linux_gnueabi
  Vsn:         1.1.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.5.0
  Type:        toolchain_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   rpi0
  MIX_ENV:      dev

|nerves_bootstrap| Environment Variable List
  target:     rpi0
  toolchain:  /Users/robert/.nerves/artifacts/nerves_toolchain_armv6_rpi_linux_gnueabi-darwin_x86_64-1.1.0
  system:     /Users/robert/.nerves/artifacts/nerves_system_rpi0-portable-1.7.2
  app:        /Users/robert/Documents/Prog/ars/ars

|nerves_bootstrap| Loadpaths End
```
* Additional information about your host, target hardware or environment that
  may help

MacOS 10.14.4, fish shell.

### Current behavior

```
$ mix deps.get
$ mix firmware
Nerves environment
  MIX_TARGET:   rpi0
  MIX_ENV:      dev

|nerves_bootstrap| Building OTP Release...
...
==> Running validation checks..
    > Mix.Releases.Checks.Erts * PASS
    > Mix.Releases.Checks.Cookie * PASS
    > Mix.Releases.Checks.LoadedOrphanedApps * PASS
==> Generated overlay vars:
    release_name=:ars
    release_version="0.1.0"
    is_upgrade=false
    upgrade_from=:latest
    dev_mode=false
    include_erts="/Users/robert/.nerves/artifacts/nerves_system_rpi0-portable-1.7.2/staging/usr/lib/erlang"
    include_src=false
    include_system_libs="/Users/robert/.nerves/artifacts/nerves_system_rpi0-portable-1.7.2/staging/usr/lib/erlang"
    erl_opts=""
    run_erl_env=""
    erts_vsn="10.3.4"
    output_dir="_build/rpi0_dev/rel/ars"
==> Copying applications to _build/rpi0_dev/rel/ars
==> Generating start_erl.data
==> Generating vm.args from rel/vm.args
==> Generating sys.config from config/config.exs
==> Including ERTS 10.3.4 from /Users/robert/.nerves/artifacts/nerves_system_rpi0-portable-1.7.2/staging/usr/lib/erlang/erts-10.3.4
==> Generating boot scripts
==> Generating RELEASES
==> Applying overlays
==> Applying mkdir overlay
    dst: releases/0.1.0/hooks
==> Applying mkdir overlay
    dst: releases/0.1.0/hooks/pre_configure.d
==> Applying mkdir overlay
    dst: releases/0.1.0/hooks/post_configure.d
==> Applying mkdir overlay
    dst: releases/0.1.0/hooks/pre_start.d
==> Applying mkdir overlay
    dst: releases/0.1.0/hooks/post_start.d
==> Applying mkdir overlay
    dst: releases/0.1.0/hooks/pre_stop.d
==> Applying mkdir overlay
    dst: releases/0.1.0/hooks/post_stop.d
==> Applying mkdir overlay
    dst: releases/0.1.0/hooks/pre_upgrade.d
==> Applying mkdir overlay
    dst: releases/0.1.0/hooks/post_upgrade.d
==> Applying copy overlay
    src: _build/rpi0_dev/lib/distillery/priv/libexec
    dst: releases/0.1.0/libexec
==> Applying mkdir overlay
    dst: releases/0.1.0/commands
==> You have strip_debug_info set to true.
    Please be aware that if you plan on performing hot upgrades later,
    this setting will prevent you from doing so without a rolling restart.
    You may ignore this warning if you have no plans to use hot upgrades.
==> Stripping release (_build/rpi0_dev/rel/ars)
==> Failed to archive release: "_build/rpi0_dev/rel/ars/lib/iex-1.8.1/ebin/Elixir.IEx.App.beam": "permission denied"
```

### Expected behavior
`Building /Users/robert/Documents/Prog/ars/ars/_build/rpi0_dev/nerves/images/ars.fw...`
After keeping debug info (`set(strip_debug_info: false)`), it works. It's not too big a deal, but I imagine there's a non-zero performance increase without debug info. The only thing I could find similar to this was maybe https://github.com/bitwalker/distillery/issues/429

JacquiManzi, 2019-07-31:
I was encountering the same problem on my macbook. Looking through that distillery thread bitwalker mentions:

> The permissions are copied directly from the source file IIRC. Can you check whether the same permissions are set on your Erlang installation, or at least on the asn1 application BEAMs? That would certainly explain why stripping them fails though. We may be able to add a check for this to Distillery, but it depends on whether we can just check the root Erlang installation directory, or if we need to check every file - the former is doable, the latter is far too expensive.

I was pretty sure that I installed Elixir with homebrew so I did a 

```
brew remove elixir
brew install elixir
```
which installed the latest 1.9 release.

I removed my deps and mix.lock file and then did a fresh 
```
mix deps.get
mix firmware
```
I needed to rename the `vm.args` file in the `rel` directory to `vm.args.eex` (because I went from Elixir 1.8.2 to 1.9.1), but the permission error coming from distillery disappeared. I _assume_ this has to do with permissions when I originally installed Elixir.

fhunleth, 2019-07-31:
@JacquiManzi Now that you're on 1.9.1, you should be using Elixir 1.9 Releases and no longer using Distillery. I'm hoping that our move to Elixir 1.9 Releases can make this issue go away. 

Certainly, if this is not true, definitely let us know.

JacquiManzi, 2019-08-01:
@fhunleth you are correct! That certainly explains why that permission error disappeared.

fhunleth, 2019-08-01:
Closing. I'm not sure how Elixir 1.9 Releases fixed this, but thankful it did. Please update or file a new issue if it returns.

closed 2019-08-01 by fhunleth