nerves-project/nerves #440: Unknown --template option for mix release.init

milmazz, 2019-07-15:
### Environment

* Elixir version (`elixir -v`): Elixir 1.9.0 (compiled with Erlang/OTP 22)
* Nerves environment: (`mix nerves.env --info`)

```console
|nerves_bootstrap| Environment Package List

  Pkg:         nerves_system_br
  Vsn:         1.8.2
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_system_rpi
  Vsn:         1.8.0
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Docker, []}

  Pkg:         nerves_toolchain_armv6_rpi_linux_gnueabi
  Vsn:         1.2.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.6.0
  Type:        toolchain_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   rpi
  MIX_ENV:      dev

|nerves_bootstrap| Environment Variable List
  target:     rpi
  toolchain:  /Users/milmazz/.nerves/artifacts/nerves_toolchain_armv6_rpi_linux_gnueabi-darwin_x86_64-1.2.0
  system:     /Users/milmazz/.nerves/artifacts/nerves_system_rpi-portable-1.8.0
  app:        /Users/milmazz/talk/bla
```

* Additional information about your host, target hardware or environment that
  may help

### Current behavior

```console
$ mix nerves.release.init
** (Mix) Could not invoke task "release.init": 1 error found!
--template : Unknown option
```

### Expected behavior

It should work as before. I just upgraded `nerves_bootstrap` to  `1.6.0`.

mobileoverlord, 2019-07-15:
`nerves.release.init` is only required when initializing distillery releases. If you are generating a new project using Elixir 1.9 you do not need to call this command.

milmazz, 2019-07-15:
@mobileoverlord I understand that for new projects `nerves` already provides a `rel/vm.args.eex`, which is suitable for releases with Elixir 1.9. But, if `nerves` plan is to forward the call from `nerves.release.init` to `release.init` that will fail for `distillery >= 2.1.0` (on that version `distillery` introduced a [breaking change](https://github.com/bitwalker/distillery/issues/694)), from now on the correct call to distillery is `mix distillery.init --template foo`.

Also, it would be helpful to mention on the _Getting Started_ guide, that `mix nerves.release.init` is no longer needed if you don't plan to use distillery in your workflow.

mobileoverlord, 2019-07-15:
Ah, I see what you are saying. I updated that PR to fix the forward call to `distillery.release.init` and also removed the bit in the getting started guides about the release configuration initialization. The getting started guides should be recommending people install elixir 1.9.0 and the new project generator will expand the new config into the mix.exs file.

milmazz, 2019-07-15:
@mobileoverlord That was fast! :) Thanks

mobileoverlord, 2019-07-31:
This is fixed in 1.5.1

closed 2019-07-31 by mobileoverlord