nerves-project/nerves #458: Better Documentation for Contributing for the Project.

TheEndIsNear, 2019-10-14:
It would be helpful to have some documentation explaining how to get started with nerves development. There isn't a lot of information currently about how to run the repo version of nerves for a project. For example, including something about setting in a separate project in the `mix.exs` file `{:nerves, path: "path/to/your/local/nerves", override: tru}` would be helpful for new developers

fhunleth, 2019-10-14:
Are you looking for using Nerves or contributing to the tooling? The reason I ask is that this is a pretty rare project for anyone to contribute to since it's tooling. Contributions here are normally documentation updates or improved error messages.

fhunleth, 2019-10-14:
In case the question is about using Nerves on a project, you can find documentation here: https://hexdocs.pm/nerves/getting-started.html. Examples can be found here: https://github.com/nerves-project/nerves_examples

TheEndIsNear, 2019-10-16:
I had more in mind for contributing to the tooling, for any kind of bugs, etc.

closed 2019-10-24 by TheEndIsNear