nerves-project/nerves #466: Nerves fails to build blinky example on RPi3

OldhamMade, 2019-11-09:
### Environment

* Elixir version (`elixir -v`): Elixir 1.9.4 (compiled with Erlang/OTP 22)
* Nerves environment: (`mix nerves.env --info`)

```
$ mix nerves.env --info
|nerves_bootstrap| Environment Package List

  Pkg:         nerves_system_br
  Vsn:         1.9.4
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_system_rpi2
  Vsn:         1.9.1
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_arm_unknown_linux_gnueabihf
  Vsn:         1.2.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.6.0
  Type:        toolchain_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   rpi2
  MIX_ENV:      dev

|nerves_bootstrap| Environment Variable List
  target:     rpi2
  toolchain:  /home/pi/.nerves/artifacts/nerves_toolchain_arm_unknown_linux_gnueabihf-linux_arm-1.2.0
  system:     /home/pi/.nerves/artifacts/nerves_system_rpi2-portable-1.9.1
  app:        /home/pi/nerves_examples/blinky

|nerves_bootstrap| Loadpaths End
```
### Current behavior

`mix firmware` fails to build blinky example despite following install instructions and installing all required deps.

```
$ MIX_TARGET=rpi2 mix firmware

Nerves environment
  MIX_TARGET:   rpi2
  MIX_ENV:      dev

==> nerves_runtime
mkdir -p /home/pi/nerves_examples/blinky/_build/rpi2_dev/lib/nerves_runtime/obj
mkdir -p /home/pi/nerves_examples/blinky/_build/rpi2_dev/lib/nerves_runtime/priv
/home/pi/.nerves/artifacts/nerves_toolchain_arm_unknown_linux_gnueabihf-linux_arm-1.2.0/bin/arm-unknown-linux-gnueabihf-gcc -c -I/home/pi/.nerves/artifacts/nerves_system_rpi2-portable-1.9.1/staging/usr/lib/erlang/erts-10.5.1/include -I/home/pi/.nerves/artifacts/nerves_system_rpi2-portable-1.9.1/staging/usr/lib/erlang/lib/erl_interface-3.13/include -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -pipe -Os -I/home/pi/.nerves/artifacts/nerves_system_rpi2-po
rtable-1.9.1/staging/usr/include -std=gnu99 -o /home/pi/nerves_examples/blinky/_build/rpi2_dev/lib/nerves_runtime/obj/nerves_runtime.o src/nerves_runtime.c
as: loadlocale.c:130: _nl_intern_locale_data: Assertion `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))' failed.
Assembler messages:
Internal error (Aborted).
Please report this bug.
src/nerves_runtime.c:22:1: fatal error: error closing -: Broken pipe
 }
 ^
compilation terminated.
make: *** [Makefile:77: /home/pi/nerves_examples/blinky/_build/rpi2_dev/lib/nerves_runtime/obj/nerves_runtime.o] Error 1
could not compile dependency :nerves_runtime, "mix compile" failed. You can recompile this dependency with "mix deps.compile nerves_runtime", update it with "mix deps.update nerves_runtime" or clean it with "mix deps.clean nerves_runtime"
==> blinky
** (Mix) Could not compile with "make" (exit status: 2).
You need to have gcc and make installed. If you are using
Ubuntu or any other Debian-based system, install the packages
"build-essential". Also install "erlang-dev" package if not
included in your Erlang/OTP version. If you're on Fedora, run
"dnf group install 'Development Tools'".
```

### Expected behavior

The blinky example should build the firmware successfully.

fhunleth, 2019-11-09:
You're building on a Pi? Cool.

This is very uncommon in our community. Most people build on their Mac, Windows, or Linux laptops. A long time ago we added support for building on an RPi, and I've been bumping the versions in the tools, but not testing.

Before I go too far with working on this, what Raspberry Pi are you building on? And do you have a Mac or Linux laptop/desktop around that you could use instead? Just to set expectations, the Raspberry Pis are pretty slow for use as development systems, and I'm not sure this will be a great experience even when it's working. A Raspberry Pi 4 might be ok.

OldhamMade, 2019-11-09:
I have a Mac, but unfortunately my final project is a kiosk and will eventually need to build on a linux machine – I've already confirmed it won't build on the Mac. 

I'll be deployed to an RPi2, but I'm using an RPi3b to build. I don't mind it being slow, as long as it works!

fhunleth, 2019-11-10:
Everything with Nerves should build on a Mac. 

For the Raspberry Pi build issue, I'd first look at https://github.com/nerves-project/toolchains/#canadian-cross-builds-for-raspberry-pi. Maybe there's an issue with how the toolchains are built? You'll need a pretty fast Linux machine to build the toolchain, though.

OldhamMade, 2019-11-10:
> Everything with Nerves should build on a Mac.
Unfortunately the kiosk library I need to use requires `qmake` and the build fails on MacOS.

I'll take a look at the link you provided and report back. 👍

fhunleth, 2020-02-26:
Closing - if there's new information, we can re-open.

closed 2020-02-26 by fhunleth