nerves-project/nerves #475: Removing erlinit overrides in the config.exs doesn't use the defaults

fhunleth, 2019-12-18:
The way to reproduce this is take a project and add the following to the `config.exs`:

```elixir
config :nerves, :erlinit,
  ctty: "anything_here"
```

Run `mix firmware` to build. The file `_build/${MIX_TARGET}_dev/nerves/rootfs_overlay/etc/erlinit.config` should exist and look like it was updated. You can also unzip the .fw file and unsquashfs the root filesystem and see that the `erlinit.config` is as it should be.

Then delete the `:erlinit` override from the `config.exs`.

Run `mix firmware`.

See that `_build/${MIX_TARGET}_dev/nerves/rootfs_overlay/etc/erlinit.config` still exists with the old configuration and that unzipping/unsquashfs'ing the .fw file still shows the old config.

The way I worked around this was to run `rm _build/${MIX_TARGET}_dev/nerves/rootfs_overlay/etc/erlinit.config` and then `mix firmware`.

mobileoverlord, 2019-12-18:
This is because the intermediate `erlinit.config` file is not being generated if the configuration is not being overridden. We should probably always generate the intermediate from the original values?

fhunleth, 2019-12-18:
I think that's a good idea. I was also thinking that there should be a comment at the top of the generated `erlinit.config` to let people know that it's a processed file. Maybe something like:

```
# Generated from rootfs_overlay/etc/erlinit.config [and overrides from the application config]
```

Or something like that so that it was easier to see where things were coming from if there was confusion.

closed 2019-12-19 by mobileoverlord