nerves-project/nerves #508: Cross compile step does not  compile all Erlang libraries

kotedo, 2020-04-06:
### Environment

* Elixir version (`elixir -v`):
Erlang/OTP 22 [erts-10.5] [source] [64-bit] [smp:4:4] [ds:4:4:10] [async-threads:1] [hipe]
Elixir 1.9.0 (compiled with Erlang/OTP 22)

* Nerves environment: (`mix nerves.env --info`)
kai@nerves-box-2:~/Bridge/relay$ MIX_TARGET=rpi3 mix nerves.env --info
|nerves_bootstrap| Environment Package List
  Pkg:         nerves_toolchain_arm_unknown_linux_gnueabihf
  Vsn:         1.3.1
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}
  Pkg:         nerves_system_rpi3
  Vsn:         1.11.0
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}
  Pkg:         nerves_toolchain_ctng
  Vsn:         1.7.1
  Type:        toolchain_platform
  BuildRunner: {nil, []}
  Pkg:         nerves_system_br
  Vsn:         1.11.0
  Type:        system_platform
  BuildRunner: {nil, []}
|nerves_bootstrap| Loadpaths Start
Nerves environment
  MIX_TARGET:   rpi3
  MIX_ENV:      dev
|nerves_bootstrap| Environment Variable List
  target:     rpi3
  toolchain:  /home/kai/.nerves/artifacts/nerves-toolchain
  system:     /home/kai/.nerves/artifacts/nerves_system_rpi3-portable-1.11.0
  app:        /home/kai/Bridge/relay
|nerves_bootstrap| Loadpaths End

* Additional information about your host, target hardware or environment that
  may help

Target is a Raspberry 3 Model 3+.
The goal is to get an application onto the PI that can deal with different databases, some of them are going to be using ODBC.  Thus, compiling the Erlang odbc library into the correct executable (ARM based) is essential.

### Current behavior
When I try to compile the firmware, the following error happens:

kai@nerves-box-2:~/Bridge/relay$ MIX_TARGET=rpi3 mix firmware
Nerves environment
  MIX_TARGET:   rpi3
  MIX_ENV:      dev

|nerves_bootstrap| Building OTP Release...
* using config/releases.exs to configure the release at runtime
* creating _build/rpi3_dev/rel/relay/releases/0.1.0/vm.args
Updating base firmware image with Erlang release...
scrub-otp-release.sh: ERROR: Unexpected executable format for '/home/kai/Bridge/relay/_build/_nerves-tmp/rootfs-additions/srv/erlang/lib/odbc-2.12.4/priv/bin/odbcserver'
Got:
 readelf:Advanced Micro Devices X86-64;0x0
Expecting:
 readelf:ARM;0x5000400, Version5 EABI, hard-float ABI
This file was compiled for the host or a different target and probably
will not work.
Check the following:
1. Are you using a path dependency in your mix deps? If so, run
   'mix clean' in that directory to avoid pulling in any of its
   build products.
2. Did you recently upgrade to Elixir 1.9 or Nerves 1.5?
   Nerves 1.5 adds support for Elixir 1.9 Releases and requires
   you to either add an Elixir 1.9 Release configuration or add
   Distillery as a dependency. Without this, the OTP binaries
   for your build machine will get included incorrectly and cause
   this error. See
   https://hexdocs.pm/nerves/updating-projects.html#updating-from-v1-4-to-v1-5
3. Did you recently upgrade or change your Nerves system? If so,
   try cleaning and rebuilding this project and its deps.
4. Are you building outside of Nerves' mix integration? If so,
   make sure that you've sourced 'nerves-env.sh'.
If you're still having trouble, please file an issue on Github
at https://github.com/nerves-project/nerves_system_br/issues.
** (Mix) Nerves encountered an error. %IO.Stream{device: :standard_io, line_or_bytes: :line, raw: true}

### Expected behavior
This step should create the firmware for the target system: rpi3

fhunleth, 2020-04-06:
Confirmed.

Adding `BR2_PACKAGE_UNIXODBC=y` to the `nerves_defconfig` fixes this. 

I think the decision is whether we add `unixodbc` to our official systems or leave it to you to make a custom system. The argument that this is part of Erlang/OTP is compelling even though I didn't think that many people used `:odbc`. 

As for the consequences, `unixodbc` adds 1 MB (uncompressed) to all firmware images made with the official systems. This seems like a less dubious add than some of the other ones we've made in the past. @mobileoverlord - any thoughts?

mobileoverlord, 2020-04-07:
My only experience with `:odbc` is dated, but memorably poor. I was attempting to use it to communicate with MSSQL, and at this time it was lacking support for "modern types" like UUID. I was always suspect of them being dated, but if it works, I think adding it would be okay. We might be able to trim it out post system build if need be.

kotedo, 2020-04-07:
The library works, even though, or maybe because it such a dated library. However, it is essential if you want to use MSSQL, Sybase and probably a few more databases.  We would be happy to see the odbc library part of Nerves’ build system.

Thank you.

Kai Janson
Relator • Ideation • Learner • Individualization • Strategic
Achiever • Futuristic • Belief • Restorative • Developer

mobileoverlord, 2020-04-07:
For MSSQL and possible sybase, you could try `https://github.com/livehelpnow/tds`

kotedo, 2020-04-07:
We might also encounter pure ODBC database connections in the (near) future.  I will be looking into TDS as well, but for now we will have to stick with Erlang’s odbc, unfortunately. The last step is to get Nerves working for our IoT devices.

Thank you.

Kai Janson
Relator • Ideation • Learner • Individualization • Strategic
Achiever • Futuristic • Belief • Restorative • Developer

fhunleth, 2020-04-07:
Ok, let me summarize:

1. @kotedo - you have a workaround (create a custom system with `BR2_PACKAGE_UNIXODBC=y`), so you should be set.
2. The OTP release code is copying host-side OTP libraries when target ones aren't found. That's an issue.
3. Do we make it possible to use `:odbc` in our official builds?

My opinion is that if the answer to #3 is no, then we must fix #2. We should probably fix #2 anyway, but I'm not sure that it applies to much else (`:wx`?).

mobileoverlord, 2020-04-07:
I added a new issue to fix #2 in what @fhunleth was talking about. @kotedo You can follow the guides [here](https://hexdocs.pm/nerves/customizing-systems.html#content) for information on how to customize a nerves system so you can enable the unix odbc, and compile the system erlang with ODBC support.

fhunleth, 2020-04-07:
@mobileoverlord So I assume that your feeling is that we do NOT include `:odbc` support in our official builds?

mobileoverlord, 2020-04-07:
Sorry, I misread your response. I am in favor of including it considering its part of erlang with the intention that we can prune out unused code when producing firmware.

fhunleth, 2020-04-07:
See https://github.com/nerves-project/nerves_system_rpi3/pull/140. Once that is merged, I'll send PRs to all official Nerves systems.

ConnorRigby, 2020-07-02:
@fhunleth this should be fixed now that the change is in the systems right?

fhunleth, 2020-07-02:
Yes. Thanks for the catch.

closed 2020-07-02 by fhunleth