nerves-project/nerves #512: 'mix firmware' fails running inside Docker container on WSL 2.

john-terrell, 2020-04-10:
### Environment

* Elixir version (`elixir -v`):
Erlang/OTP 22 [erts-10.6.4] [source] [64-bit] [smp:12:12] [ds:12:12:10] [async-threads:1] [hipe]
Elixir 1.10.2 (compiled with Erlang/OTP 22)

* Nerves environment: (`mix nerves.env --info`)
|nerves_bootstrap| Environment Package List

  Pkg:         nerves_system_br
  Vsn:         1.11.0
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_system_nemo2
  Vsn:         2.6.0
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_arm_unknown_linux_gnueabihf
  Vsn:         1.3.1
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.7.1
  Type:        toolchain_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start

* Additional information about your host, target hardware or environment that
  may help

NOTE: Running inside a docker container that's hosted inside WSL 2 (this is new-ish Docker-Desktop feature)

### Current behavior
1) When building, system attempts to use fwup.exe instead of fwup.  Looks like it needs to detect it's running inside a docker container?   I can create a symbolic link from fwup to fwup.exe to make this work.

2) mix firmware.burn also fails with an error that it's attempting to run cmd.exe.   This will fails because it's running inside a Docker container on WSL.

### Expected behavior
When running inside a Docker container, I'd expect that WSL specific behavior inside Nerves would be disabled.   (It's unclear of the WSL behavior inside Nerves is only meant for WSL v1?)

john-terrell, 2020-04-10:
Here's the error returned when trying to run 'mix firmware.burn'

** (ErlangError) Erlang error: :enoent
    (elixir 1.10.2) lib/system.ex:795: System.cmd("cmd.exe", ["/c", "echo %TEMP%"], [])
    lib/nerves/utils/wsl.ex:94: Nerves.Utils.WSL.get_temp_file_location/1
    lib/nerves/utils/wsl.ex:237: Nerves.Utils.WSL.make_file_accessible/3
    lib/mix/tasks/burn.ex:66: Mix.Tasks.Burn.run/1
    (mix 1.10.2) lib/mix/task.ex:330: Mix.Task.run_task/3
    (mix 1.10.2) lib/mix/cli.ex:82: Mix.CLI.run_task/2

fhunleth, 2020-04-11:
Thanks for the error report. I think that you may be one of our first WSL 2 users.

@GregMefford or @michaelkschmidt - do you have any suggestions?

mobileoverlord, 2020-04-11:
I have a windows 10 VM set up with wsl2 and I’ve found that using Ubuntu command line works. I am unfamiliar with your setup. Do you have docker install instructions for WSL2 that I could try to replicate your env?

john-terrell, 2020-04-12:
Sure, see here: https://docs.docker.com/docker-for-windows/wsl-tech-preview/

fhunleth, 2020-11-14:
Closing since this issue seems to have gone away with the non-preview version of WSL2. If this is wrong, then please let us know and we can try to figure out what is different about our setups.

closed 2020-11-14 by fhunleth