nerves-project/nerves #531: Mount writable filesystem to `/data`

fhunleth, 2020-07-27:
Currently the writable filesystem is mounted to the home directory of the root user, `/root`. This seems to cause some confusion since the word "root" is used in so many places.

One way around this would be to change the mount point to be `/data` and add a symlink from `/root` to `/data` to avoid breaking programs. Then we could update all of the docs and when talking about were to put writable data, we could say that it should be put in the `/data` directory. That sounds less confusing. I think this also gives us an opportunity to make the language around the "application data partition" consistent.

I'm open to other words as well. `/data` is what I've seen on embedded Linux systems using read-only root filesystems, but there might be better words. I don't know how widespread the use of `/data` for writable filesystem in A/B embedded Linux systems is either.

ConnorRigby, 2020-07-27:
Android uses `/data` for it's data partition as well if i remember correctly.  I'm all for this.

mobileoverlord, 2020-07-28:
- I agree; I believe people get confused by `/root`. I like `/data`. Maybe we could symlink them for compatibility?

fhunleth, 2020-10-03:
The current systems have a symlink from `/data` to `/root`. That's probably the best we can do without breaking existing projects for now.

closed 2020-10-03 by fhunleth