nerves-project/nerves #557: Erlinit template doesn't work with elixir 1.11

ConnorRigby, 2020-10-13:
### Environment

* Elixir version (`elixir -v`):
```
Elixir 1.11.0 (compiled with Erlang/OTP 23)
```

### Current behavior

Configuring erlinit in `config.exs` does nothing on elixir 1.11

```elixir
config :nerves,
  erlinit: [
    ctty: "ttyAMA0",
    hostname_pattern: "nerves-%s"
  ]
```
This exact code will work correctly on elixir 1.10 and older

ConnorRigby, 2020-10-13:
This is actually the same bug as https://github.com/phoenixframework/phoenix/issues/4026

fhunleth, 2020-10-13:
Are there any clues as to what's wrong? I'm guessing that one of us is going to have to dig into this or at least come up with a simple example to encourage anyone else to help us debug it. I'm not sure I'll have time today to look at it any, but maybe tomorrow evening.

josevalim, 2020-10-20:
Is this happening in an umbrella/poncho app? Is the configuration replicated in all config files that may potentially be interested in erlinit?

ConnorRigby, 2020-10-20:
this particular bug has no umbrella/poncho stuff going on. `Application.get_env` simply returns `nil` on stuff that is configured in application env.

ConnorRigby, 2020-10-20:
[elixir-1.11-bug.zip](https://github.com/nerves-project/nerves/files/5410528/elixir-1.11-bug.zip)

I modified this file https://github.com/nerves-project/nerves/blob/ea8c59f56a43c48cf3cd45c5c745fef9fa81e407/lib/mix/tasks/firmware.ex#L228 locally and added:

```elixir
    Mix.raise """
    Application env: #{inspect(Application.get_all_env(:nerves))}
    """
``` 
at the top of the function, and i get:

```
env MIX_TARGET=rpi3 mix firmware                               8.8s  Tue 20 Oct 2020 10:41:55 AM PDT
==> nerves
/usr/bin/make -C src all
make[1]: Entering directory '/home/connor/workspace/bug/erlinit_bug/deps/nerves/src'
cc /home/connor/workspace/bug/erlinit_bug/_build/rpi3_dev/lib/nerves/obj/port.o  -o /home/connor/workspace/bug/erlinit_bug/_build/rpi3_dev/lib/nerves/priv/port
make[1]: Leaving directory '/home/connor/workspace/bug/erlinit_bug/deps/nerves/src'
if [ -f test/fixtures/port/Makefile ]; then /usr/bin/make -C test/fixtures/port; fi
Compiling 1 file (.ex)
==> erlinit_bug

Nerves environment
  MIX_TARGET:   rpi3
  MIX_ENV:      dev

Compiling 2 files (.ex)
Generated erlinit_bug app
|nerves_bootstrap| Building OTP Release...

* skipping runtime configuration (config/runtime.exs not found)
* creating _build/rpi3_dev/rel/erlinit_bug/releases/0.1.0/vm.args
** (Mix) Application env: []
```

Even tho the env is certainly configured.

josevalim, 2020-10-20:
The env is configured on config/target.exs though and MIX_TARGET != host, which means it won't be loaded? Or am I missing something?

josevalim, 2020-10-20:
Oh, it is the opposite, MIX_TARGET!=host, which means it will be loaded. Hrm...

ConnorRigby, 2020-10-20:
Correct. I initially thought it was because `:nerves` is tagged as `runtime: false` but that didn't seem to be the problem.

ConnorRigby, 2020-10-20:
I tentatively suspect the Nerves tooling at this point having something to do with it, however i'm not exactly sure **why**. I can't seem to reproduce this outside of Nerves.

josevalim, 2020-10-20:
The issue is with the loadconfig alias. For some reason, it is not invoking loadconfig at all, so none of the config files are loaded.

josevalim, 2020-10-20:
I found the elixir commit that broke it.

ConnorRigby, 2020-10-20:
FWIW, i believe we have been playing with removing that alias.

josevalim, 2020-10-20:
Fixed in Elixir master and v1.11 branch.

Rewriting the alias to do this:

    aliases: [loadconfig: [&bootstrap/1, "loadconfig"]]

where bootstrap only starts the app should also work.

fhunleth, 2020-10-20:
Super news!!! Thanks @josevalim!!!

closed 2020-10-21 by fhunleth