nerves-project/nerves #559: `mix firmware` no such option 'delta-source-raw-offset'

mattlenz, 2020-10-16:
Running through the nerves setup tutorial, I get the following when I try to build the firmware bundle;

```
Building <project_path>/_build/rpi3_dev/nerves/images/<project_name>.fw...
/home/matt/.nerves/artifacts/nerves_system_rpi3-portable-1.13.0/images/fwup.conf:479: no such option 'delta-source-raw-offset'
fwup: Error parsing configuration file '/home/matt/.nerves/artifacts/nerves_system_rpi3-portable-1.13.0/images/fwup.conf'
** (Mix) Nerves encountered an error. %IO.Stream{device: :standard_io, line_or_bytes: :line, raw: true}
```

I thought my fwup version v1.5.1 may be an issue, so I've also tried 1.5.0, 1.5.2, same error.

Can anyone point me in the right direction?

### Environment

* Elixir version (`elixir -v`): 1.10.4
* Nerves environment: (`mix nerves.env --info`)

```
==> nerves
/usr/bin/make -C src all
make[1]: Entering directory '<project_path>/deps/nerves/src'
cc <project_path>/_build/dev/lib/nerves/obj/port.o  -o <project_path>/_build/dev/lib/nerves/priv/port
make[1]: Leaving directory '<project_path>/deps/nerves/src'
if [ -f test/fixtures/port/Makefile ]; then /usr/bin/make -C test/fixtures/port; fi
==> <project_name>
|nerves_bootstrap| Environment Package List

  No packages found
|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   host
  MIX_ENV:      dev

NERVES_SYSTEM is unset
NERVES_TOOLCHAIN is unset
|nerves_bootstrap| Environment Variable List
  target:     host
  toolchain:  unset
  system:     unset
  app:        <project_path>

|nerves_bootstrap| Loadpaths End
```

jjcarstens, 2020-10-16:
@mattlenz You'll need the minimum fwup version of `1.6`. The latest version is `1.8.1`. Can you try a fwup version `>= 1.6` and see if that resolves this?

mattlenz, 2020-10-16:
Hi @jjcarstens -- oh duh! Just installed fwup 1.8.1 from source and looks like we're rolling!

Sorry, I assumed the fwup in the arch user repository was more recent.

closed 2020-10-16 by mattlenz