nerves-project/nerves #562: Nerves.Artifact.BuildRunners.Docker failure: missing skeleton files

basvanwesting, 2020-10-29:
### Environment

```
➜  elixir -v
Erlang/OTP 23 [erts-11.0.4] [source] [64-bit] [smp:8:8] [ds:8:8:10] [async-threads:1] [hipe]

Elixir 1.10.4 (compiled with Erlang/OTP 23)
```

```
➜  custom_rpi4_project mix nerves.env --info
==> nerves
==> custom_rpi4_project
|nerves_bootstrap| Environment Package List

  Pkg:         custom_rpi4
  Vsn:         1.13.0
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Docker, []}

  Pkg:         nerves_system_br
  Vsn:         1.13.2
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_toolchain_arm_unknown_linux_gnueabihf
  Vsn:         1.3.2
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.7.2
  Type:        toolchain_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   custom_rpi4
  MIX_ENV:      dev

Local Nerves system detected but is not compiled:

  custom_rpi4

|nerves_bootstrap| Environment Variable List
  target:     custom_rpi4
  toolchain:  /Users/basvanwesting/.nerves/artifacts/nerves_toolchain_arm_unknown_linux_gnueabihf-darwin_x86_64-1.3.2
  system:     /Users/basvanwesting/.nerves/artifacts/custom_rpi4-portable-1.13.0
  app:        /Users/basvanwesting/sandbox/custom_rpi4_project

|nerves_bootstrap| Loadpaths End
```

```
➜  docker -v
Docker version 19.03.13, build 4484c46d9d

➜  uname -a
Darwin [...] 18.7.0 Darwin Kernel Version 18.7.0: Mon Aug 31 20:53:32 PDT 2020; root:xnu-4903.278.44~1/RELEASE_X86_64 x86_64
```

### Current behavior

I'm following the steps in https://hexdocs.pm/nerves/customizing-systems.html, adjusted for nerves-system v1.13.0 and rpi4.
No deviations from the docs:

https://github.com/basvanwesting/custom_rpi4.git
https://github.com/basvanwesting/custom_rpi4_project.git

[... 138 lines cut ...]
tar: system/skeleton/dev/stderr: Cannot utime: No such file or directory
tar: system/skeleton/dev/fd: Cannot utime: No such file or directory
tar: package/skeleton-init-sysv/skeleton/var/tmp: Cannot utime: No such file or directory
tar: package/skeleton-init-sysv/skeleton/var/spool: Cannot utime: No such file or directory
tar: package/skeleton-init-sysv/skeleton/var/run: Cannot utime: No such file or directory
tar: package/skeleton-init-sysv/skeleton/var/log: Cannot utime: No such file or directory
tar: package/skeleton-init-sysv/skeleton/var/lock: Cannot utime: No such file or directory
tar: package/skeleton-init-sysv/skeleton/var/lib/misc: Cannot utime: No such file or directory
tar: package/skeleton-init-sysv/skeleton/var/cache: Cannot utime: No such file or directory
tar: package/skeleton-init-sysv/skeleton/dev/log: Cannot utime: No such file or directory
tar: package/skeleton-init-openrc/skeleton/var/tmp: Cannot utime: No such file or directory
tar: package/skeleton-init-openrc/skeleton/var/spool: Cannot utime: No such file or directory
tar: package/skeleton-init-openrc/skeleton/var/run: Cannot utime: No such file or directory
tar: package/skeleton-init-openrc/skeleton/var/log: Cannot utime: No such file or directory
tar: package/skeleton-init-openrc/skeleton/var/lock: Cannot utime: No such file or directory
tar: package/skeleton-init-openrc/skeleton/var/lib/misc: Cannot utime: No such file or directory
tar: package/skeleton-init-openrc/skeleton/var/cache: Cannot utime: No such file or directory
tar: package/skeleton-init-openrc/skeleton/etc/runlevels/default/sysv-rcs: Cannot utime: No such file or directory
tar: package/skeleton-init-openrc/skeleton/dev/log: Cannot utime: No such file or directory
tar: Exiting with failure status due to previous errors
```

### Expected behavior

No failure to compile.

mobileoverlord, 2020-10-29:
Hi, thanks for the report, its curious that its missing access to those files specifically. I know its bin a bit since some of us have used the docker build runner, but its nice to have for running on OS X natively. I can take a peek into why this is failing over the next couple days. In the meantime, if you have access to a linux machine or a VM, you can compile your system artifact there.

takasehideki, 2020-10-29:
Hi! I have also encountered this issue. The solution at that time is to uncheck (disable) "Use gRPC FUSE for file sharing" in Preference section of Docker Desktop for Mac.

basvanwesting, 2020-10-29:
Thanks @takasehideki, that does fix it for me!

basvanwesting, 2020-10-29:
I'll close the issue when the build has completed, since it is more of a Docker Desktop issue I think.

basvanwesting, 2020-10-29:
The build completed successfully, so I'm closing this issue.

@mobileoverlord, I've run into issues with Docker on macOS before, so a separate Linux machine which can run docker natively is probably a good idea in general. I did try the same custom nerves-system-rpi4 approach as described above on a Raspberry Pi 4 with Ubuntu 20.0.4 64bit as well. But it complained about needing a "64-bit Linux for cross-compiling", so I went the Docker macOS route, assuming the toolchain wasn't meant to run on ARM64. Maybe I should post a separate issue for that? What do you think?

closed 2020-10-29 by basvanwesting