nerves-project/nerves #566: MQTT connection to AWS IoT with Rpi0

MAkcanca, 2020-11-29:
### Environment

* Elixir version:  Erlang/OTP 23 [erts-11.1.3] [source] [64-bit] [smp:2:2] [ds:2:2:10] [async-threads:1] [hipe]
* Nerves environment: rpi0

### Current behavior

I'm trying to create an example project with AWS IoT Core, and it's supposed to just connect to the MQTT server and send hello world message, using the tortoise library. I even did a code search on whole github but couldn't find a proper working example. I believe AWS IoT and Nerves could have a happy marriage but it's really hard to bring them together (at least for me).
I wrote a simple connection function using the resources(error questions) I found from here and there. This code hangs, probably due to SSL keychain, depth, hostname_check, sni or other mambo jambo. I tried certifi.cacerts, certifi.cacertsfile etc but had no success. Is it really that hard to just send a Hello World message to an AWS IoT MQTT server using Nerves?
```
        client_opts = [ 
            client_id: "testdev01",
            handler: {Tortoise.Handler.Logger, []},
            server: {Tortoise.Transport.SSL,
            cacerts: :certifi.cacerts(),
            certfile: "/srv/erlang/lib/network_led-0.1.0/priv/testdev01.cert",
            keyfile: "/srv/erlang/lib/network_led-0.1.0/priv/testdev01.private.key",
            depth: 99,
            host: 'axxxxxxxxxx-ats.iot.eu-central-1.amazonaws.com',
            port: 8883,
            customize_hostname_check: [match_fun: :public_key.pkix_verify_hostname_match_fun(:https)],
            versions: [:'tlsv1.2'],
            server_name_indication: '*.iot.eu-central-1.amazonaws.com',
            },
            subscriptions: [{"#", 0}]
        ]
        Tortoise.Connection.start_link(client_opts)
        Tortoise.publish("testdev01", "sensor/test", "hello")
```

### Expected behavior
It should publish the MQTT message with proper settings. An example implementation could help a lot.

I also raised this issue on Tortoise library. https://github.com/gausby/tortoise/issues/127

Thanks a lot.

fhunleth, 2020-11-29:
@MAkcanca - You're not alone. MQTT to AWS IoT is surprisingly hard to debug. It's not really the MQTT part, but getting the SSL part right. My company has been working with @gausby to try to make this easier by adding a library around Tortoise called Jackalope. See https://github.com/smartrent/jackalope. Unfortunately, the most important code snippet for you was dropped from the repo. I added https://github.com/smartrent/jackalope/issues/28 to see if someone could post it back. I'll check back during the week if no one responds.

closed 2020-11-29 by fhunleth

jjcarstens, 2020-11-29:
FWIW, first time communication with AWS MQTT typically requires that your signer CA cert be in with the `cacerts` list as well.

So for your options, you'd need something like:

```elixir
...
    cacerts: [your_signer_ca_der | :certifi.cacerts()],
    certfile: "/srv/erlang/lib/network_led-0.1.0/priv/testdev01.cert",
    keyfile: "/srv/erlang/lib/network_led-0.1.0/priv/testdev01.private.key",
...
```

where `your_signer_ca_der` is the CA certificate used to create `testdev01.cert`, read in and converted to DER format.

This is typically just needed for initial connect. If that doesn't fix things, there is probably something with the AWS setup that needs to be handled there

MAkcanca, 2020-11-29:
> FWIW, first time communication with AWS MQTT typically requires that your signer CA cert be in with the `cacerts` list as well.
>
> So for your options, you'd need something like:
>
> ```elixir
> ... [...]

I will try this one out and update here. Thank you for the insight!

MAkcanca, 2020-11-30:
@jjcarstens How can I read in the der file? I tried giving just the path, and using File.read(path) but It gives `SSL WARNING: Ignoring a CA cert as it could not be correctly` error. What is the correct way to do that?

jjcarstens, 2020-11-30:
```elixir
singer_cert_path
|> File.read!()
|> X509.Certificate.from_pem!()
|> X509.Certificate.to_der()
```
This should prob do it for you

MAkcanca, 2020-11-30:
What is the correct way to read in the cert file as der?
> ```elixir
> singer_cert_path
> |> File.read!()
> |> X509.Certificate.from_pem!()
> |> X509.Certificate.to_der()
> ```
> 
> This should prob do it for you

MAkcanca, 2020-11-30:
I added in the AmazonRootCA1 since I used their One Click Certificate to generate certs, but it still hangs.