nerves-project/nerves #577: PIZero W Unable to boot if compiled as rpi target  (no gadget mode) with nerves 1.7 (OK with 1.6.5)

cocoaway, 2020-12-26:
### Environment

* Elixir version (`elixir -v`): 1.11.2-otp-23

* Nerves environment: (`mix nerves.env --info`) 1.7.1
*|nerves_bootstrap| Loadpaths Start
Nerves environment
  MIX_TARGET:   rpi
  MIX_ENV:      dev

|nerves_bootstrap| Environment Variable List
  target:     rpi
  toolchain:  /.nerves/artifacts/nerves_toolchain_armv6_rpi_linux_gnueabi-darwin_x86_64-1.3.2
  system:     /.nerves/artifacts/nerves_system_rpi-portable-1.13.2

* Additional information about your host, target hardware or environment that
  may help
MacOS as host.
Vintagenet configured as access point.

If compiled using nerves version 1.6.5 and elixir 1.10 , PiZero boots perfectly and ssh connection is availabe.

### Current behavior

NO BOOT

### Expected behavior

fhunleth, 2020-12-26:
Are you able to connect your PiZero to a monitor via the HDMI port? There could be additional information that's printed out there.

cocoaway, 2020-12-27:
I first need to retrieve an hdmi cable with adapter ...
---
Reproducing the problem is very simple because it is based on hello_nerves in getting started.

mix nerves.new hello_nerves

in hello_nerves

export MIX_TARGET = rpi

mix deps.get

In /config/target.exs added wlan0 as access point in: vintage_net

config :vintage_net,
  regulatory_domain: "US",
  config: [
    #{"usb0", %{type: VintageNetDirect}},
    #{"eth0",
    # %{
    #   type: VintageNetEthernet,
    #   ipv4: %{method: :dhcp}
    # }},
    {"wlan0", %{type: VintageNetWiFi,
                vintage_net_wifi: %{
                  networks: [
                    %{
                      mode: :ap,
                      ssid: "pitest",
                      key_mgmt: :none
                    }
                  ]
                  },
                ipv4: %{
                  method: :static,
                  address: "192.168.24.1",
                  gateway: "192.168.24.255",
                  netmask: "255.255.255.0",
                  name_servers: ["1.1.1.1"]
                },
                dhcpd: %{
                  start: "192.168.24.2",
                  end:   "192.168.24.5"
                }
                }
    }
  ]

[image]

mix deps.get
mix firmware.burn

Insert SD card in rpi0 W , it starts the boot but after few led bilnks all freeze, led off and no wifi ap.

fhunleth, 2020-12-27:
I'm able to reproduce the issue. 

This is a Broadcom/Cypress WiFi firmware problem and the patch that was put into `nerves_system_rpi` to fix it was slightly wrong. A workaround that works ok in the US regulatory domain is to go to your project and create a `rootfs_overlay/lib/firmware/brcm` directory. Then put 
[brcmfmac43430-sdio.raspberrypi,model-zero-w.txt](https://github.com/nerves-project/nerves/files/5744664/brcmfmac43430-sdio.raspberrypi.model-zero-w.txt) into it. Rebuild the firmware and try it again. 

I'm going to fix this properly so that you won't need that `brcmfmac43430-sdio.raspberrypi,model-zero-w.txt` file.

Thank you for reporting this!!!

fhunleth, 2020-12-27:
Update to my previous comment: Adding the `.txt` file alone doesn't fix everything. The next error that I see is "no clm_blob available" for the WiFi driver. I'm investigating.

fhunleth, 2020-12-27:
Could you also remove the `gateway: "192.168.24.255",` line? That's the other fix. It works for me now.

This is interesting that the gateway specification line broke the configuration as bad as it did. I'll see if I can add some checks for that to `vintage_net`. 

Let me know if adding the `rootfs_overlay/lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt` and removing `gateway: "192.168.24.255",` from your `target.exs` fix the issue for you. Thanks!

cocoaway, 2020-12-27:
I will try to add your patch and test it . for completeness of information I had tried `regulatory_domain: "IT" ` too, with the same result.

cocoaway, 2020-12-27:
Done as you suggested, created` rootfs_overlay/lib/firmware/brcm/ ` and copied` brcmfmac43430-sdio.raspberrypi.model-zero-w.txt`  , removed `gateway: "192.168.24.255"` from target.exs,  but no boot.

cocoaway, 2020-12-27:
I've seen this in compiler output : 

`Pseudo modify file "lib/firmware/brcm/brcmfmac43430a0-sdio.ONDA-V80" does not exist in source filesystem.  Ignoring.`

fhunleth, 2020-12-27:
That `ONDA-V80` print is something else. I get that too. I'm going to fix it, but it won't cause any issues.

fhunleth, 2020-12-27:
Sorry, I don't have any other ideas. I just tried my setup again. Could you try again? If it still doesn't work, could you post a link to your firmware image (the .fw file - run `find . -name "*.fw"` to find it) and I'll double check that the changes took.

cocoaway, 2020-12-28:
Hi, something is changed.
I recompiled and restarted. The LED turns on and off very little, remains on and almost always off. However, I saw the "pitest" access point appear for a few seconds, then it disappears to reappear again after a few minutes. It is as if the pizero continues to boot after boot.

I was able to connect just in time to download the boot log.

Attached the log file.
[PI0RingLogger.txt](https://github.com/nerves-project/nerves/files/5746264/PI0RingLogger.txt)

here is the link to fw : https://we.tl/t-82xM8so5kS

Thank you.

cocoaway, 2020-12-28:
Probably the most interesting lines : 

00:00:13.206 [info]  brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1

00:00:14.868 [info]  brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1

00:00:14.868 [info]  **brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2)**, device may have limited channels available

00:00:14.869 [info]  brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3b33decd

I've found this related topic : https://github.com/raspberrypi/linux/issues/3659 , maybe it can help.

cocoaway, 2020-12-30:
Hi Frank,
now it seems OK !

The pizero green led is alway off, but wifi AP is available.

Attached the log.

[PI0RingLogger_2.txt](https://github.com/nerves-project/nerves/files/5753462/PI0RingLogger_2.txt)

fhunleth, 2020-12-30:
Sorry - I've been away for the holidays. I'm glad it works. I'll merge the fix to `nerves_system_rpi0` for the missing wifi firmware txt file fix. This won't affect you, but at some point you'll be able to delete `rootfs_overlays/lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi.model-zero-w.txt`. I also fixed the `ONDA-V80` non-issue so next update that should go away too.

Have a good end of year and new year!

closed 2020-12-30 by fhunleth

cocoaway, 2020-12-31:
Hi, there is a new.

Yesterday I've tested with `Nerves 1.7,` compiled with` elixir 1.11.2-otp-23 `and` erlang 23.1.4.` :  deleted ` gateway` and `name_servers` from `target.exs`

It compiled ok and is running without any issue.

The AP always remains operational without disappearing and then reappearing.

All this **without** inserting the` rootfs_overlays / lib / firmware / brcm / brcmfmac43430-sdio.raspberrypi.model-zero-w.txt file`

If you need the fw image, I'm here.

Thank you.