nerves-project/nerves #579: Uploading a firmware patch fails while writing rootfs.img to a device

kentaro, 2020-12-30:
I tried the new feature called [firmware patches](https://hexdocs.pm/nerves/experimental-features.html#firmware-patches) and got an error described below.

### Environment

* Elixir version (`elixir -v`): Elixir 1.11.2 (compiled with Erlang/OTP 23)
* Nerves environment (`mix nerves.env --info`): details are hidden below
<details>
<pre>
==> nerves
==> target_app_patch
|nerves_bootstrap| Environment Package List

  Pkg:         nerves_system_br
  Vsn:         1.13.5
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_system_rpi3
  Vsn:         1.13.2
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Docker, []}

  Pkg:         nerves_toolchain_arm_unknown_linux_gnueabihf
  Vsn:         1.3.2
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.7.2
  Type:        toolchain_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   rpi3
  MIX_ENV:      dev

|nerves_bootstrap| Environment Variable List
  target:     rpi3
  toolchain:  /Users/antipop/.nerves/artifacts/nerves_toolchain_arm_unknown_linux_gnueabihf-darwin_x86_64-1.3.2
  system:     /Users/antipop/.nerves/artifacts/nerves_system_rpi3-portable-1.13.2
  app:        /Users/antipop/src/github.com/kentaro/deployment-methods-evaluation/target_app_patch

|nerves_bootstrap| Loadpaths End
</pre>
</details>

* Additional information about your host, target hardware or environment that
  may help:
  * fwup version: 1.8.3 (1.8.2 also reproduces the problem)

### Current behavior

I got a firmware patch by executing `mix firmware.patch`.

```
Finished generating patch firmware

Source
/Users/antipop/src/github.com/kentaro/deployment-methods-evaluation/target_app_patch/_build/rpi3_dev/nerves/images/target_app_patch.fw
uuid: 3307f71a-97dc-59f6-767b-6f870596c6d5"

size: 43698827 bytes

Target
/Users/antipop/src/github.com/kentaro/deployment-methods-evaluation/target_app_patch/_build/rpi3_dev/nerves/images/patch/target.fw
uuid: 8da26d54-d121-5ca9-450c-82a293df1f24"

size: 43990202 bytes

Patch
/Users/antipop/src/github.com/kentaro/deployment-methods-evaluation/target_app_patch/_build/rpi3_dev/nerves/images/patch.fw
size: 40172772 bytes
```

However, I encountered the error below when I attempted to upload the patch by executing `mix upload` as below:

```
mix upload --firmware _build/rpi3_dev/nerves/images/patch.fw
==> nerves
==> target_app_patch

Nerves environment
  MIX_TARGET:   rpi3
  MIX_ENV:      dev

Path: /Users/antipop/src/github.com/kentaro/deployment-methods-evaluation/target_app_patch/_build/rpi3_dev/nerves/images/patch.fw

Uploading to nerves.local...

fwup: Upgrading partition A
fwup: raw_write wrote 68268032 bytes for 'rootfs.img', but should have written 39297024
** (Mix) ssh failed with status 1
```

### Expected behavior

`rootfs.img` is successfully written to the device by executing `mix upload`.

fhunleth, 2020-12-30:
Interesting. I don't think that we have good tooling to debug firmware patches, so this may be hard to debug. I'll try.

First off, firmware patches really don't work well with dev builds due to debug data interfering with patch generation. I've never put any time into seeing if this is possible to fix. My suggestion would be to switch to prod builds or update your `mix.exs` so that it strips releases. I have been putting `strip_beams: [keep: ["Docs"]]` in some of my projects since I use the online docs way more that the Erlang debug data. This obviously won't fix your problem, though.

Is `target_app_patch.fw` a patch or a full firmware file? It needs to be the full firmware file for whatever is running on the device. The error that you received makes me this that it is right, but I never added a check for this case, so I'm wondering if the code could have been tricked.

Could you switch to a temp directory and unzip `patch.fw` and post the meta.conf file? I'd also like to see the `meta.conf` inside `target.fw`? If these are just projects created with `mix nerves.new`, then perhaps I can reproduce. (I've only used firmware patches that I've manually created and through NervesHub)

@mobileoverlord Do you have any other ideas? I skimmed the `mix firmware.patch` code again and it looks like what I do manually...

jjcarstens, 2020-12-30:
There might be a few issues relating to order of operations here

1. what does your `fwup.conf` look like? Does it have the required changes for delta offsets set as described in https://hexdocs.pm/nerves/experimental-features.html#preparing-your-nerves-system
2. if your fwup.conf has those needed changes, was that before or after initially burning firmware to the SD card

For 1, the offsets for delta changes needs to be set and correct for your use. If not, it will fail.

I'm assuming you do have that set, so next is 2 - if you made an initial firmware without any changes for delta and are trying to upload after making the fwup changes, that won't work. The existing fw on device doesn't know the needed information to make appropriate assertions on offsets. You would need to have made the fwup changes, burned the SD, then made patch and attempted upload

kentaro, 2020-12-30:
Thank you for your quick and kind reply!

@jjcarstens 

`mix nerves.new` seemed to generate those required config into `deps/nerves_system_rpi3/fwup.conf` as below:

```
task upgrade.a {
...
    on-resource rootfs.img {
        delta-source-raw-offset=${ROOTFS_B_PART_OFFSET}
        delta-source-raw-count=${ROOTFS_B_PART_COUNT}
        raw_write(${ROOTFS_A_PART_OFFSET})
    }
...
}
```

```
task upgrade.b 
    on-resource rootfs.img {
        delta-source-raw-offset=${ROOTFS_A_PART_OFFSET}
        delta-source-raw-count=${ROOTFS_A_PART_COUNT}
        raw_write(${ROOTFS_B_PART_OFFSET})
    }
}
```

To confirm your hypothesis, I once again made a local change and burn it to an SD card as a full firmware. Then I inject the card into my rpi3 device, made another change, and upload the change as a firmware patch via LAN cable. Again I got the same error I reported above.

@fhunleth 

> Is `target_app_patch.fw` a patch or a full firmware file?

It's a full firmware file I guess (the file name looks confusing, though).

> I'd also like to see the `meta.conf` inside `target.fw` ?

I found a weird thing related to it. As we see, the output I pasted in the above comment said:

```
Target
/Users/antipop/src/github.com/kentaro/deployment-methods-evaluation/target_app_patch/_build/rpi3_dev/nerves/images/patch/target.fw
uuid: 8da26d54-d121-5ca9-450c-82a293df1f24"
```

However, there's no such file. I didn't remove it,  of course.

```
ls -la /Users/antipop/src/github.com/kentaro/deployment-methods-evaluation/target_app_patch/_build/rpi3_dev/nerves/images/patch/target.fw
ls: /Users/antipop/src/github.com/kentaro/deployment-methods-evaluation/target_app_patch/_build/rpi3_dev/nerves/images/patch/target.fw: No such file or directory
```

Anyway, here's the `meta.conf` retrieved from `patch.fw`: (details are hidden)
<detail>
```
meta-product="target_app_patch"
meta-version="0.1.0"
meta-author="The Nerves Team"
meta-platform="rpi3"
meta-architecture="arm"
require-fwup-version="0.15.0"
file-resource "bootcode.bin" {
length=52480
blake2b-256="d2f5865b405330d1495540fcc82946f40e902176f25be75ac2d9bd4caf8befe7"
}
file-resource "fixup.dat" {
length=10275
blake2b-256="dc9d6e33833049bfeb57408ec45afcc615aed924c4a0c9fe756c0ebda2a1fed0"
}
[... 248 lines cut ...]
funlist={"3","fat_write","38693","overlays/ramoops.dtbo"}
}
on-resource "rootfs.img" {
delta-source-raw-offset="77324"
delta-source-raw-count=289044
funlist={"2","raw_write","366368"}
}
}
task "upgrade.unexpected" {
reqlist={"4","require-uboot-variable","uboot-env","a.nerves_fw_platform","rpi3","4","require-uboot-variable","uboot-env","a.nerves_fw_architecture","arm"}
on-init {
funlist={"2","error","Please check the media being upgraded. It doesn't look like either the A or B partitions are active."}
}
}
task "upgrade.wrongplatform" {
on-init {
funlist={"2","error","Expecting platform=rpi3 and architecture=arm"}
}
}
uboot-environment "uboot-env" {
block-offset=16
block-count=16
}
```
</detail>

fhunleth, 2021-01-01:
FYI - I posted a "fix" that technically fixes this issue. However, my opinion that this mix task is trying to be too smart with the patches and is going to be confusing at best. The main problem is that the firmware on the device is guessed to be the same one created by `mix firmware`. However, once you patch the device once, that guess is wrong and the firmware image running on the device isn't available for the next patch. What needs to happen is that firmware images are cached and the upload logic can refer back to the cache for how to create an appropriate patch. NervesHub does this automatically, but we don't have anything that does this locally. I have a partially complete change that will remove the guesses from `mix firmware.patch`. This will make it unfriendly to use, but my hope is that it will make it understandable and usable by someone wanting to make a friendly firmware patch uploader.

kentaro, 2021-01-02:
Thank you for informing me of your opinion! I haven't fully gotten what you mean because of my lack of understanding of the gut of `mix firmware.patch`. However, I think it can be better.

kentaro, 2021-01-02:
#580 "technically" fixed this issue. Thanks!

closed 2021-01-02 by kentaro

kentaro, 2021-01-02:
I got the error while uploading a firmware patch for the second time using the current head of nerves after #580 merged.

```
$ mix upload --firmware _build/rpi3_dev/nerves/images/patch.fw
==> nerves
make[1]: Nothing to be done for `all'.
==> target_app_patch

Nerves environment
  MIX_TARGET:   rpi3
  MIX_ENV:      dev

Path: /Users/antipop/src/github.com/kentaro/deployment-methods-evaluation/target_app_patch/_build/rpi3_dev/nerves/images/patch.fw

Uploading to nerves.local...

fwup: Upgrading partition A
fwup: xdelta3 error: target window checksum mismatch
** (Mix) ssh failed with status 1
```

Is this because of what you commented above?

> However, once you patch the device once, that guess is wrong and the firmware image running on the device isn't available for the next patch.

fhunleth, 2021-01-02:
Yes, you can't run it twice. It's not usable for what I think that you're doing. I'll try to finish off that code change that I was talking about so that `mix firmware.patch` will make more sense. Right now, I think `mix firmware.patch` is only valuable to see the result of tweaking firmware generation options on patch size.

kentaro, 2021-01-03:
Thank you! I got it!

PR #580 (2021-01-02): Fix target/source swap when generating firmware patches
Fixes #579
commit 8ad47607 (2021-01-02): Fix target/source swap when generating firmware patches

Fixes #579