nerves-project/nerves #581: erlang native :timer fails in Nerves

cocoaway, 2021-01-04:
[ :timer ](https://erlang.org/doc/man/timer.html#module)native library functions [apply_after/4](https://erlang.org/doc/man/timer.html#apply_after-4) , [apply_interval/4](https://erlang.org/doc/man/timer.html#apply_interval-4) fail in Nerves.

The functions are accepted but not executed.

This fail in Nerves Iex : 

>:timer.apply_interval(1000, IO, :puts, ["weeeee"])
{:ok, {:interval, #Reference<0.4076832015.805830657.133553>}}
 IO.puts("weeeee") is not executed every second. Is not executed at all.

 no output
 Expected  > weeee

in macos iex
> :timer.apply_interval(1000, IO, :puts, ["weeeee"])
{:ok, {:interval, #Reference<0.162820322.2933391363.5362>}}
weeeee
weeeee
weeeee
weeeee
weeeee
weeeee
weeeee
...

This fails in Nerves :  
>:timer.apply_after(1000, IO, :puts, ["weeeee"])
{:ok, {6081038803, #Reference<0.4076832015.805830657.133677>}}

nothing happen, but  "weeeee" is expected to be printed.

but not in macos iex
> :timer.apply_after(1000, IO, :puts, ["weeeee"])
{:ok, {-576460749227118, #Reference<0.324260181.249298948.171860>}}
> weeeee

os_monotonic_resolution 
in Nerves 
> :erlang.system_info(:os_monotonic_time_source)
[
  function: :clock_gettime,
  clock_id: :CLOCK_MONOTONIC,
  resolution: 1000000000,
  extended: :no,
  parallel: :yes,
  time: 7402019820000
]

in macos.
> :erlang.system_info(:os_monotonic_time_source)
[
  function: :clock_gettime,
  clock_id: :CLOCK_MONOTONIC,
  resolution: 1000000,
  extended: :no,
  parallel: :yes,
  time: 520816687401000
]

Include errors, stacktraces, screenshots or code that may help us reproduce the
issue.

### Environment

* Elixir version (`elixir -v`):
 Erlang/OTP 23 [erts-11.1.3] [source] [64-bit] [smp:4:4] [ds:4:4:10] [async-threads:1] [hipe]
 Elixir 1.11.2 (compiled with Erlang/OTP 23)

* Nerves environment: (`mix nerves.env --info`)
*acme
|nerves_bootstrap| Environment Package List

  Pkg:         nerves_system_br
  Vsn:         1.13.5
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_system_rpi
  Vsn:         1.13.2
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Docker, []}

  Pkg:         nerves_toolchain_armv6_rpi_linux_gnueabi
  Vsn:         1.3.2
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.7.2
  Type:        toolchain_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   rpi
  MIX_ENV:      dev

|nerves_bootstrap| Environment Variable List
  target:     rpi
  toolchain:  /Users/xx/.nerves/artifacts/nerves_toolchain_armv6_rpi_linux_gnueabi-darwin_x86_64-1.3.2
  system:     /Users/xx/.nerves/artifacts/nerves_system_rpi-portable-1.13.2

* Additional information about your host, target hardware or environment that
  may help

macOS Catalina
asdf 
elixir          1.11.2-otp-23
erlang       23.1.4

### Current behavior

Include errors, stacktraces, screenshots or code that may help us reproduce the
issue.

### Expected behavior

fhunleth, 2021-01-04:
This works for me on a Nerves device:

```
iex(5)> :timer.apply_interval(1000, IO, :puts, ["weeeee"])
{:ok, {:interval, #Reference<0.3430611437.1342177281.202220>}}
iex(6)> weeeee
weeeee
weeeee
weeeee
weeeee
weeeee
```

I suspect that you're printing somewhere that you're not looking. Prints go to the console associated with the process running the code. The `:timer` callback is happening in a different process. I assume that you've connected over ssh. The startup console will get the prints since the `:timer` processes are started early on. These come out the UART in my setup, but would be the HDMI output by default for you. 

Try this instead:

```
:timer.apply_interval(1000, Logger, :bare_log, [:info, "weee"])
```

Then look at the log via `RingLogger.next` or `RingLogger.attach`.

```
iex(6)> :timer.apply_interval(1000, Logger, :bare_log, [:info, "weee"])
{:ok, {:interval, #Reference<0.2471941791.805306369.47410>}}
iex(7)> RingLogger.next                                                

17:37:06.184 [info]  weee

17:37:07.184 [info]  weee

2 out of 2 new messages shown.
```

closed 2021-01-04 by fhunleth

cocoaway, 2021-01-04:
Thank you Frank !