nerves-project/nerves #582: windows support without wsl

typedefstructer, 2021-01-07:
Is it possible to get windows support without wsl/wls2, wls2 settings messes up my virtualbox vms.

If its not currently possible, I would like to try and make it happen(at least just for my PC)
,so please point me in the right direction

thanks

fhunleth, 2021-01-07:
We don't build crosscompilers for native Windows so Nerves won't work without WSL. While you could update our toolchain support to generate Windows native versions, that requires a fast Linux box to be practical. I think that your best bet is to install Ubuntu in a VM and develop with Nerves in there.

closed 2021-01-07 by fhunleth

typedefstructer, 2021-01-07:
@fhunleth 

got it but is there a way just to get this done in my pc, 
and maybe anyone who wants this flexibility, can follow some steps to get this done just for their system locally.

if you can give me a high level list of things needed to be done to get the native support, it'll be great.

Also for clarification, when you said it requires a fast linux box, are you talking about automated builds or something else...?