nerves-project/nerves #583: Arch Linux: Could not find application file: nerves_runtime.app

chrisdambrosio, 2021-01-12:
### Environment

* Elixir version (`elixir -v`):
```
Erlang/OTP 23 [erts-11.1.3] [source] [64-bit] [smp:8:8] [ds:8:8:10] [async-threads:1] [hipe]

Elixir 1.11.0 (compiled with Erlang/OTP 23)
```
* Nerves environment: (`mix nerves.env --info`)
```
==> nerves
==> app
|nerves_bootstrap| Environment Package List

  No packages found
|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   host
  MIX_ENV:      dev

NERVES_SYSTEM is unset
NERVES_TOOLCHAIN is unset
|nerves_bootstrap| Environment Variable List
  target:     host
  toolchain:  unset
  system:     unset
  app:        /app

|nerves_bootstrap| Loadpaths End
```
* Additional information about your host, target hardware or environment that
  may help

Reproduce with a docker container:

```
docker run -it --rm archlinux:base-20210110.0.13332 bash -c '
  pacman -Syy --noconfirm
  pacman-db-upgrade
  pacman -S --noconfirm --needed base-devel ncurses5-compat-libs openssh-askpass git squashfs-tools curl
  pacman -S --noconfirm --needed gcc make elixir
  mix local.hex --force
  mix local.rebar --force
  mix archive.install hex nerves_bootstrap --force
  yes | mix nerves.new app
  cd app
  mix deps.get
  mix test
  bash
'
```

### Current behavior

In Arch Linux, after creating a new project with `mix nerves.new`, and starting the nerves app by running the tests (`mix test`) or starting the REPL (`iex -S mix`) the error message is returned:
```
** (Mix) Could not start application nerves_runtime: could not find application file: nerves_runtime.app
```
I've tested w/ MacOS and Ubuntu and I am not seeing the error there.

### Expected behavior

The tests run and the REPL is started without the error message.

ConnorRigby, 2021-01-12:
i don't think this is specific to arch linux. (i run arch btw :laughing: ). i think this is part of the new project generator.

chrisdambrosio, 2021-01-12:
@ConnorRigby do you mean that it is specific to the project generator under Arch? Because I am not seeing the error for projects generated on Mac or Ubuntu.

fhunleth, 2021-01-13:
@chrisdambrosio I can reproduce the issue. I don't know why it's happening yet, but it's definitely there. Thanks for the report and for working out an easy way to reproduce with Docker since I don't use arch Linux.

mobileoverlord, 2021-01-13:
Could you try using elixir 1.11.3 and see if you still experience the issue?

chrisdambrosio, 2021-01-13:
@mobileoverlord Ok, so I installed 1.11.3 with asdf and that does resolve the issue.

I can use that for now, thanks.

closed 2021-01-13 by mobileoverlord