nerves-project/nerves #612: Suggestion of creating a default community health file

pojiro, 2021-04-24:
I found dead links in nerves_sytem_br which linked to CODE_OF_CONDUCT.md or CONTRIBUTING.md in this repo.

At first I considered fixing the links, but then I thought that those markdown files would be common to the nerves project.
So, I suggest to create common markdowns for the nerves project.

The way to do this may be found in the following link.
https://docs.github.com/en/communities/setting-up-your-project-for-healthy-contributions/creating-a-default-community-health-file

fhunleth, 2021-04-24:
Thanks! I didn't know about that it was possible to make a default community health file. That sounds much better.

I just created https://github.com/nerves-project/.github and added the community files here. I didn't add the issue template, since I was less sure that made sense.

If I understand correctly, the process now is to delete the community files in every other repository and fix the links? See #613 for this repository. If you think that I did this right and no one else on the Nerves core team sees an issue, I'll merge and then we can go from there with all of the other repositories.

pojiro, 2021-04-24:
> the process now is to delete the community files in every other repository and fix the links?

Yes, and I think #613 is right.

The downside of this way is that the locally cloned repository will no longer have health markdowns.
Please consider if that fits the policy of this project.

fhunleth, 2021-04-24:
> The downside of this way is that the locally cloned repository will no longer have health markdowns.
> Please consider if that fits the policy of this project.

I don't have a problem with that. The important part to me is that the contributing guidelines and code of conduct appear during interactions on GitHub. We don't distribute these in the hex packages.

Thank you for bringing this up. I've noticed that links to the files have now started showing up on Nerves projects where we forgot to add the files, so this is an improvement already.

closed 2021-04-24 by fhunleth