nerves-project/nerves #623: The end-user project is always recompiled when MIX_TARGET is set

axelson, 2021-06-11:
### Environment

* Elixir version (`elixir -v`): Elixir 1.9.4 (compiled with Erlang/OTP 22)
* Nerves environment:
<details>
<summary>mix nerves.env --info</summary> 

==> nerves
/usr/bin/make -C src all
make[1]: Entering directory '/home/jason/dev/forks/nerves_examples/blinky/deps/nerves/src'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/jason/dev/forks/nerves_examples/blinky/deps/nerves/src'
if [ -f test/fixtures/port/Makefile ]; then /usr/bin/make -C test/fixtures/port; fi
==> blinky
|nerves_bootstrap| Environment Package List

  Pkg:         nerves_system_br
  Vsn:         1.11.4
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_system_rpi3
  Vsn:         1.11.2
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_arm_unknown_linux_gnueabihf
  Vsn:         1.3.2
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.7.2
  Type:        toolchain_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   rpi3
  MIX_ENV:      dev

|nerves_bootstrap| Environment Variable List
  target:     rpi3
  toolchain:  /home/jason/.local/share/nerves/artifacts/nerves_toolchain_arm_unknown_linux_gnueabihf-linux_x86_64-1.3.2
  system:     /home/jason/.local/share/nerves/artifacts/nerves_system_rpi3-portable-1.11.2
  app:        /home/jason/dev/forks/nerves_examples/blinky

|nerves_bootstrap| Loadpaths End

</details>

* Additional information about your host, target hardware or environment that
  may help

### Current behavior

Reproduction repo (includes steps)
https://github.com/axelson/nerves_examples/tree/test-regression

### Expected behavior

Project is only recompiled once when running `mix compile` (unless changes are made of course)

axelson, 2021-06-11:
I have confirmed that this regression was caused by: https://github.com/nerves-project/nerves/commit/40a77d8a9511fa9f14b0c5204ebda3c3ce0d3c76

axelson, 2021-06-11:
A weird thing is that if I use nerves as a `:path` dependency ec5ae3f2639ededd229ee159d5ec9aa8e23b5470 then the project still always gets recompiled, but without the make output.

<details>
<summary>
MIX_DEBUG=true mix compile --verbose # with nerves :path dep pointing to good commit (does recompile but no make output)
</summary>

** Running mix loadconfig (inside Blinky.MixProject)
** Running mix compile --verbose (inside Blinky.MixProject)
** Running mix loadpaths --verbose (inside Blinky.MixProject)
** Running mix archive.check --verbose (inside Blinky.MixProject)
** Running mix nerves.loadpaths (inside Blinky.MixProject)
** Running mix nerves.precompile (inside Blinky.MixProject)
** Running mix deps.precompile (inside Blinky.MixProject)
==> nerves
** Running mix compile --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.MixProject)
** Running mix loadpaths --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.MixProject)
** Running mix compile.all --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.MixProject)
** Running mix compile.yecc --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.MixProject)
** Running mix compile.leex --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.MixProject)
** Running mix compile.erlang --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.MixProject)
** Running mix compile.elixir --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.MixProject)
** Running mix compile.xref --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.MixProject)
** Running mix compile.app --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.MixProject)
==> nerves_system_br
** Running mix compile --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.System.BR.Mixfile)
** Running mix loadpaths --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.System.BR.Mixfile)
** Running mix compile.all --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.System.BR.Mixfile)
** Running mix compile.app --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.System.BR.Mixfile)
==> nerves_toolchain_ctng
** Running mix compile --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.Toolchain.Ctng.MixProject)
** Running mix loadpaths --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.Toolchain.Ctng.MixProject)
** Running mix compile.all --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.Toolchain.Ctng.MixProject)
** Running mix compile.yecc --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.Toolchain.Ctng.MixProject)
** Running mix compile.leex --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.Toolchain.Ctng.MixProject)
** Running mix compile.erlang --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.Toolchain.Ctng.MixProject)
** Running mix compile.elixir --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.Toolchain.Ctng.MixProject)
** Running mix compile.xref --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.Toolchain.Ctng.MixProject)
** Running mix compile.app --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.Toolchain.Ctng.MixProject)
==> nerves_system_linter
** Running mix compile --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.System.Linter.Mixfile)
** Running mix loadpaths --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.System.Linter.Mixfile)
** Running mix compile.all --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.System.Linter.Mixfile)
** Running mix compile.yecc --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.System.Linter.Mixfile)
** Running mix compile.leex --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.System.Linter.Mixfile)
** Running mix compile.erlang --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.System.Linter.Mixfile)
** Running mix compile.elixir --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.System.Linter.Mixfile)
** Running mix compile.xref --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.System.Linter.Mixfile)
** Running mix compile.app --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside Nerves.System.Linter.Mixfile)
==> nerves_toolchain_arm_unknown_linux_gnueabihf
** Running mix compile --no-deps --no-archives-check --no-elixir-version-check --no-warnings-as-errors (inside NervesToolchainArmUnknownLinuxGnueabihf.MixProject)
[... cut ...]
** Running mix compile.all --verbose (inside Blinky.MixProject)
** Running mix compile.yecc --verbose (inside Blinky.MixProject)
** Running mix compile.leex --verbose (inside Blinky.MixProject)
** Running mix compile.erlang --verbose (inside Blinky.MixProject)
** Running mix compile.elixir --verbose (inside Blinky.MixProject)
** Running mix compile.xref --verbose (inside Blinky.MixProject)
** Running mix compile (inside Blinky.MixProject)
** Running mix compile.app --verbose (inside Blinky.MixProject)
</details>

axelson, 2021-06-11:
So it looks like the main issue is that `elixir_make` always returns `{:ok, []}` ([source link](https://github.com/elixir-lang/elixir_make/blob/2016764ff1ce9cb2fe9fa88404eaac9612234690/lib/mix/tasks/compile.make.ex#L153)), which is indicating that `nerves` was "recompiled", which means `mix will_recompile` is run on the user's application which forces recompilation of all modules.

So it seems that we would need to teach `elixir_make` how to detect if anything was actually compiled or not. Which in the general case is most likely not determinable. But I believe it might be possible by creating a specification for the makefile, or perhaps nerves could detect that no files are changed and skip running the elixir_make compiler.

axelson, 2021-06-12:
Okay, the above info is correct, but I was thrown off for a bit because `:path` dependencies are treated rather differently when it comes to detecting if recompilation is needed.

Here is a Proof of Concept workaround for this issue:
https://github.com/axelson/elixir_make/commit/3c618a799885568fd51562cddbd3ee1c043fd640

But since all it does is *always* return `:noop`, it is not a solution

*But* here is a potential working approach :tada::
https://github.com/axelson/elixir_make/commit/fc5b0bf6a520e2eedae2b25143ab530df43ae348

There's some relevant comments and questions inline and I would appreciate any feedback before submitting it as a PR to elixir_make

Here is the commit that uses the patched version of elixir_make that I used to verify that the PoC fix works
https://github.com/axelson/nerves_examples/commit/45b1d63cf5c50da3b95a43b637561ecb3b751f23

Here is the commit that uses the potential working fix in blinky:
https://github.com/axelson/nerves_examples/commit/b793671c8ee8dc39ee61b047507af481c7c452ae

That's a lot of links! Hopefully I didn't mess up any of them :smiley:

fhunleth, 2021-06-12:
Thanks for all of the debug. My only idea so for is to add an option to `:elixir_make` to specify an output file to check. `:elixir_make` would check the file before calling `make` and afterwards. If the file has the same timestamp before and afterwards, then return `:noop`.