nerves-project/nerves #632: Support Config Providers

axelson, 2021-07-01:
### Environment

* Elixir version (`elixir -v`): Elixir 1.12.1 (compiled with Erlang/OTP 24)
* Nerves environment: 
<details>
<summary>mix nerves.env --info</summary>
jason@jdesktop ~/d/v/fw (next-release-hex)> mix nerves.env --info
==> nerves
==> fw
|nerves_bootstrap| Environment Package List

  Pkg:         nerves_system_br
  Vsn:         1.16.0
  Type:        system_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.8.3
  Type:        toolchain_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_system_vultr
  Vsn:         0.13.0
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_x86_64_nerves_linux_musl
  Vsn:         1.4.2
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

|nerves_bootstrap| Loadpaths Start

Nerves environment
  MIX_TARGET:   vultr
  MIX_ENV:      prod

|nerves_bootstrap| Environment Variable List
  target:     vultr
  toolchain:  /home/jason/.local/share/nerves/artifacts/nerves_toolchain_x86_64_nerves_linux_musl-linux_x86_64-1.4.2
  system:     /home/jason/.local/share/nerves/artifacts/nerves_system_vultr-portable-0.13.0
  app:        .

|nerves_bootstrap| Loadpaths End
</details>
* Additional information about your host, target hardware or environment that
  may help
  * deploying to vultr.com cloud compute server

### Current behavior

When a Config.Provider is added to the project then there is a message that the environment variable `RELEASE_SYS_CONFIG` could not be fetched:
[image]
Note that the line in the output that starts with `provider (provider.ex:219)` is a line that I added to elixir for debugging

### Expected behavior

The config provider runs successfully and the config it returns is merged.

### Thoughts

`RELEASE_SYS_CONFIG` is supposed to point to the sys.config file which does exist, so a possible fix could be to simply set the environment file. But I'm not sure where in nerves it should be set and I'm not sure if there's a good reason not to set it.

Relevant docs:
> RELEASE_SYS_CONFIG - the location of the sys.config file. It can be set to a custom path and it must not include the .config extension
https://hexdocs.pm/mix/master/Mix.Tasks.Release.html

I have not yet tried to reproduce this on other nerves systems, but I suspect that the same error would appear.

fhunleth, 2021-07-01:
Support would need to be added to `erlinit`, but I wonder why Elixir can't figure this out itself since the `sys.config` is in the release that it makes.

Out of curiosity, what would be a use for a config provider?

axelson, 2021-07-01:
My use case for the config provider is to store secrets on the device in a file instead of in the git repository. And potentially to even use something like etcd in the future.

fhunleth, 2021-07-01:
I realized that erlinit already lets you set arbitrary environment variables, so you can do this today:

```elixir
config :nerves, :erlinit, env: "RELEASE_SYS_CONFIG=path"
```

axelson, 2021-07-03:
Thanks! I just confirmed that:

```elixir
config :nerves, :erlinit, env: "RELEASE_SYS_CONFIG=/srv/erlang/releases/0.1.0/sys"
```
(although with the downside that if I update my app's version I'll have to update this configuration line)

fhunleth, 2021-07-06:
It's possible for `erlinit` to add this line automatically. 

I'm still confused as to why `RELEASE_SYS_CONFIG` is needed, since it seems like it should be possible to figure out automatically. There aren't any requirements for the directory to be writable, are there? 

I suppose that even if there are caveats to `erlinit` setting this automatically, the caveats aren't as confusing as the current state where it isn't set at all. Right?

axelson, 2021-07-08:
> I'm still confused as to why `RELEASE_SYS_CONFIG` is needed, since it seems like it should be possible to figure out automatically. There aren't any requirements for the directory to be writable, are there?

I haven't looked into the details of the release process to be knowledgeable to answer this, but I wouldn't think that the directory needs to be writeable.

> I suppose that even if there are caveats to `erlinit` setting this automatically, the caveats aren't as confusing as the current state where it isn't set at all. Right?

I don't forsee any caveats (famous last words!) and I would expect that a user would be able to overwrite the `RELEASE_SYS_CONFIG` setting similar to how I showed above if it is necessary for some reason.

axelson, 2021-08-10:
@fhunleth I did not have to add `RELEASE_ROOT`, although I'm using this with https://github.com/nerves-project/nerves_system_vultr
The only relevant change I have is:
```
 config :nerves,
   erlinit: [
-    hostname_pattern: "nerves-%s"
+    hostname_pattern: "nerves-%s",
+    # Workaround for https://github.com/nerves-project/nerves/issues/632
+    env: "RELEASE_SYS_CONFIG=/srv/erlang/releases/0.1.0/sys"
   ]
```

fhunleth, 2021-08-10:
Thanks. Then https://github.com/nerves-project/erlinit/pull/88 will fix the issue that you ran into. It will be in `nerves_system_br 1.16.4`.

closed 2021-08-10 by fhunleth

axelson, 2021-08-12:
Fantastic, thank you!