nerves-project/nerves #679: Adjust burn task to detect if drive already has Nerves image

jjcarstens, 2021-12-15:
By default `mix burn` (and subsequently `mix firmware.burn`) will completely overwrite the SD card because it defaults to using the `complete` task, losing all data on it

However, it may be possible to detect the SD card already has an image on it and adjust the task to use `upgrade` instead which would preserve the `/data` partition on disk like when upgrading OTA via SSH or NervesHub

axelson, 2021-12-15:
Oooh, that would be nice! Another option would be for `mix firmware.burn` to not allow overwriting an existing nerves image and instead require `mix firmware.burn --overwrite` to actually overwrite the image.

jjcarstens, 2021-12-15:
That's precisely what I was thinking @axelson! Default to `upgrade`, which would fail if there is no Uboot env (according to Frank), then fallback to complete. And also allow `--overwrite` option to force it