nerves-project/nerves #683: Path and other errors

yaniyaniyaniyaniyaniyani, 2022-01-02:
`All dependencies are up to date

Nerves environment
  MIX_TARGET:   host
  MIX_ENV:      dev

==> nerves

Microsoft (R) Program Maintenance Utility Version 14.30.30706.0
Copyright (C) Microsoft Corporation.  All rights reserved.

NMAKE : fatal error U1052: file 'Makefile.win' not found
Stop.
could not compile dependency :nerves, "mix compile" failed. Errors may have been logged above. You can recompile this dependency with "mix deps.compile nerves", update it with "mix deps.update nerves" or clean it with "mix deps.clean nerves"
==> hello_nerves
** (Mix) Could not compile with "nmake" (exit status: 2).
One option is to install a recent version of
[Visual C++ Build Tools](https://visualstudio.microsoft.com/visual-cpp-build-tools/)
either manually or using [Chocolatey](https://chocolatey.org/) -
`choco install VisualCppBuildTools`.

After installing Visual C++ Build Tools, look in the "Program Files (x86)"
directory and search for "Microsoft Visual Studio". Note down the full path
of the folder with the highest version number. Open the "run" command and
type in the following command (make sure that the path and version number
are correct):

    cmd /K "C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\vcvarsall.bat" amd64

This should open up a command prompt with the necessary environment variables
set, and from which you will be able to run the "mix compile", "mix deps.compile",
and "mix test" commands.

Another option is to install the Linux compatiblity tools from [MSYS2](https://www.msys2.org/).

After installation start the msys64 bit terminal from the start menu and install the
C/C++ compiler toolchain. E.g.:

  pacman -S --noconfirm pacman-mirrors pkg-config
  pacman -S --noconfirm --needed base-devel autoconf automake make libtool git \
    mingw-w64-x86_64-toolchain mingw-w64-x86_64-openssl mingw-w64-x86_64-libtool

This will give you a compilation suite nearly compatible with Unix' standard tools.`

I suspect the issue here is that the code is referring to an older version of VS that lived in the Program File (x86) folder.

Doubt this is good practice. No VS no worky.

fhunleth, 2022-01-02:
I don't think any one has used Nerves on Windows like this. I only know of people using it on Docker, in a VM, or under WSL2. Could you try one of those ways?

yaniyaniyaniyaniyaniyani, 2022-01-02:
Yea, I'll get to putting it on something not Windows in the next few months. I'm sure there is a way to get this to use a different compiler, I was looking at it today. 

[{:elixir_make, "~> 0.4", runtime: false}]
AND
compilers: [:elixir_make] ++ Mix.compilers,

Too much of a newbie to make that work without time going in it seems.

I used the defaults and it all works good. I'm less sure about these apps loading into Windows program spaces. Erlang is in "Program Files", Phoenix in "Program Files (x86), Elixir in Program Files, I thought I notice something load into the "user" directory. Seemed a bit "where they hell was that loaded"?

Felt like everything connected to Erlang should have been in the Erlang directory. 

Not complaining... just giving you some feedback of "first contact". 

Not sure why Phoenix loaded in the ...x86 directory when it uses Erlang which is in the main Program Files? Stuff that loads in there has an "oh is this old" feeling about it.

closed 2022-01-02 by fhunleth