nerves-project/nerves #711: Firmware fails on RPi0w.

raisch, 2022-03-07:
### Environment

#### Host Hardware
Macbook Pro (16-inch 2019)
Processor 2.4GHz 8-Core Intel Core i9
Memory 64 GB 2667 MHz DDR4

#### Host Operating System
maxOS Catalina
Version 10.15.7 (19H1615)

#### Software

Elixir 1.13.3

Erlang/OTP 24 [erts-12.2.1] [source] [64-bit] [smp:16:16] [ds:16:16:10] [async-threads:1] [jit] [dtrace]

|nerves_bootstrap| Environment Package List

  Pkg:         nerves_toolchain_armv6_nerves_linux_gnueabihf
  Vsn:         1.5.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.8.5
  Type:        toolchain_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_system_rpi0
  Vsn:         1.18.2
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Docker, [make_args: ["source", "all", "legal-info"]]}

  Pkg:         nerves_system_br
  Vsn:         1.18.4
  Type:        system_platform
  BuildRunner: {nil, []}

|nerves_bootstrap| Loadpaths Start
Nerves environment
  MIX_TARGET:   rpi0
  MIX_ENV:      dev
|nerves_bootstrap| Environment Variable List
  target:     rpi0
  toolchain:  /Users/-/.nerves/artifacts/nerves_toolchain_armv6_nerves_linux_gnueabihf-darwin_x86_64-1.5.0
  system:     /Users/-/.nerves/artifacts/nerves_system_rpi0-portable-1.18.2
  app:        .
|nerves_bootstrap| Loadpaths End

#### Target Hardware: 

Raspberry Pi Zero W V1.1

SD card: Sandisk Edge 32GB

### Current behavior

After burning Nerves to SD, the Zero appears to start booting only to stop, blinking its green LED twice in succession, again and again. This occurs only with Nerves, as other SDs with Raspbian or other opsys work without a hitch.

Assuming it had something to do with my app, I created a "plain_old_nerves" app via:

- mix nerves.new ./plain_old_nerves
- cd ./plain_old_nerves
- mix deps.get
- mix firmware
- mix burn

and received the same result: No love.

### Expected behavior

Nerves to work on RPi Zero W v1.1

ConnorRigby, 2022-03-07:
I just tried it out to be sure, and it worked fine on my mac. 
I would check your network settings and see if you have a device that isn't configured incorrectly of something. 
the double blink you're seeing means its working correctly. 

Also be sure to double check you're using the correct USB port, only one of them carries data.

As a workaround, you could configure wifi for the device.

raisch, 2022-03-08:
Thanks for your reply, Connor, but I'm not sure how to proceed.

When you say "I just tried it out to be sure, and it worked fine on my
mac.", what did you try? Did you burn Nerves onto a fresh 32GB Sandisk SD
card and insert it into an RPi0 W v. 1.1?

As I mentioned in my error report, I am burning the SD card with Nerves, so
I'm not sure why you would indicate there might be a networking issue.

I have a RPi zero W v. 1.1 with an HDMI cable and usb keyboard connected to
the correct connectors, and am using the power supply and USB keyboard/hub
that came with the zero.

When I insert the newly formatted and burned SD card and power on the
device, the green led flickers once or twice before resolving into a
repeating pattern of two flashes. The HDMI monitor screen is blank
throughout and there is no new device present on my wireless LAN. If I
connect the device to my laptop via the USB cable, there is no indication
that there is any connection.

I understand that this two green flash pattern indicates that the RPi
cannot read from the boot partition on the SD card.

I have two identical RPi0 and this problem happens on both no matter which
SD card I use, and if I load the SD with Raspian, everything works like a
charm.

So, without any output available to me other than the green LED indicating
that the device cannot read from the boot partition on the SD, no matter
which SD card I use, occurs on either of two brand-new RPi0, and the device
performs as expected when I burn another firmware image to the SD, I am
left with only one conclusion.

What am I missing? And more importantly, how should I proceed?

Thanks.
--
Rob Raisch, Internet Handyman

jjcarstens, 2022-03-08:
@raisch There are a few things going on here. 
* The extra USB port is configured as OTG and can instead be used as networking from your host, but not as input so the keyboard will not work
* [HDMI output is disabled](https://github.com/nerves-project/nerves_system_rpi0/blob/main/rootfs_overlay%2Fetc%2Ferlinit.config#L13) by default and the console is directed to the UART pints instead (since the USB port cannot be used with a keyboard)
* This is raspberry pi hardware, but it is not raspian so LED sequencing is not the same. The [LED behavior on the rpi0 is changed to be `heartbeat`](https://github.com/nerves-project/nerves_system_rpi0/blob/main/config.txt#L48-L51), which is the pattern that you are describing indicating that it has booted

So everything looks to be working as expected and expectations just need to be adjusted here. There are a few things that you can do:
* plug the rpi0 into your machine via the OTG USB port (not the power USB). That should start networking between your host and device. You should then be able to `ping nerves.local` and even `ssh nerves.local`
* If you wish to use the HDMI and keyboard, build a new firmware with [`nerves_system_rpi`](https://hex.pm/packages/nerves_system_rpi) (`MIX_TARGET=rpi`) - This is virtually the same system, but HDMI is enabled and the extra USB port can be used with a keyboard

raisch, 2022-03-08:
Excellent. Thank you very much Jon. Following your instructions, I was able to 1. network to the device over USB, 2. use the HDMI output, and 3. load my current app (a simulation of the state-machine in a home security basestation) without a hitch. 
The only remaining issue for me (which is completely moot now that I have network access to the zero) is that connecting a keyboard via USB still doesn't work. In fact, if I pull the USB cable out after boot and reinsert it, the entire device reboots. But as I said, I don't require the keyboard. Thanks again and please feel free to close the ticket.

closed 2022-03-17 by jjcarstens

fhunleth, 2022-03-17:
The RPi0 reboots if you plug a USB keyboard into it and remove it?????? That's unexpected, but I've honestly never tried that. I'll try and reproduce to see what's happening.

fhunleth, 2022-03-17:
As luck would have it, I can't reproduce. The USB port is forced to gadget mode, so nothing should happen and that's what I see at the iex prompt. I suppose this isn't an issue, so I'll just let it be.