nerves-project/nerves #755: Add `:nerves_env` option support

jjcarstens, 2022-05-30:
Similar to the `make_env` option of `elixir_make` - a map of environment variables to set or function callback to be run as the environment is setting

Nerves tooling starts early enough in the process that this could be run as Nerves is initiated, but before other apps are compiled and opens the door for easy cross-compilation setup support (especially for tools like rustler or zigler)

fhunleth, 2022-05-30:
We already support injecting cross compilation environment variables via the Nerves system. How is this different?

jjcarstens, 2022-05-30:
I think that helps for the case where you have control of the system and have system specific env vars to shove in at the global level. But I don't think that aids users of the systems in being able to inject more (correct me if I'm wrong)

I'm more thinking of the situation where a user might need to set an env var, but it's dependent on what tooling and systems are currently in use (which nerves sets). This [article on using rustler with Nerves](https://cocoa-research.works/2022/01/use-rustler-with-nerves-cross-compile/) is what is really driving this idea. They are trying to read Nerves vars in the Mix project in order to set some vars which reference the current system in user before their app starts to compile.

But the way our tooling is setup, that article above only works if it is in a mix project of a dependency but not your root mix project, which is totally inconsistent behavior.

The workaround right now is you have to alias one of the nerves tasks to add a step to put your env stuff in while having access to the Nerves env

```
aliases: ["nerves.loadpaths": ["nerves.loadpaths", &make_rust_env/1]]
```

Or we potentially add some callback that can get called after we've set Nerves env vars and let others add more according to the target being compiled for

jjcarstens, 2022-05-30:
So I dug around the systems more and I think all the pieces are in there that they would need to create those env vars needed in a different way (like via a make file or env file or something)

But...I also think this option is an easy add for helping this niche situation

fhunleth, 2022-05-30:
That's really not a good workaround, and I'd recommend against using it. The Nerves system environment variables we expose that should be used are in the [docs](https://hexdocs.pm/nerves/compiling-non-beam-code.html#target-cpu-arch-os-and-abi). Trying to derive values from `NERVES_TARGET_TRIPLE` can easily lead to wrong choices. You can see this in the Raspberry Pi Zero's `TARGET_*` settings. See https://github.com/nerves-project/nerves_system_rpi0/blob/main/mix.exs#L56-L61. For the Pi Zero, it's important to set the CPU to `arm1176jzf_s`, but that information isn't in the target triple.

[Rustler Precompiled](https://github.com/philss/rustler_precompiled/blob/main/lib/rustler_precompiled.ex#L346-L355) does the right thing.

I don't support adding this option. If the `TARGET_*` environment variables are missing some information, then we should add them there since that's the only sane place for maintaining how to crosscompile. However, since Rustler Precompiled works, I don't think it's needed for this.

closed 2022-05-30 by fhunleth