nerves-project/nerves #852: Broken packages in Nix-shell file

simonelnahas, 2023-03-14:
### Environment

MacOS 13.2.1

### Current behavior

Using the file provided in the [docs](https://hexdocs.pm/nerves/installation.html#for-nixos-or-nix-package-manager) here saved as default.nix:
```nix
{ pkgs ? import <nixpkgs> {} }:

with pkgs;

mkShell {
  name = "nervesShell";
  buildInputs = [
    autoconf
    automake
    curl
    erlangR25
    fwup
    git
    pkgs.beam.packages.erlangR25.elixir
    rebar3
    squashfsTools
    x11_ssh_askpass
    pkg-config
  ];
  shellHook = ''
    SUDO_ASKPASS=${pkgs.x11_ssh_askpass}/libexec/x11-ssh-askpass
  '';
}

```
I get the following errors:
```bash
~/D/D/insta_billboard [1]> nix-shell default.nix 
error: Package ‘x11-ssh-askpass-1.2.4.1’ in /nix/store/1x89jxh81wyh2ngc18ny0figv9lkb0my-nixpkgs/nixpkgs/pkgs/tools/networking/x11-ssh-askpass/default.nix:30 is marked as broken, refusing to evaluate.

       a) To temporarily allow broken packages, you can use an environment variable
          for a single invocation of the nix tools.

            $ export NIXPKGS_ALLOW_BROKEN=1

        Note: For `nix shell`, `nix build`, `nix develop` or any other Nix 2.4+
        (Flake) command, `--impure` must be passed in order to read this
        environment variable.

       b) For `nixos-rebuild` you can set
         { nixpkgs.config.allowBroken = true; }
       in configuration.nix to override this.

       c) For `nix-env`, `nix-build`, `nix-shell` or any other Nix command you can add
         { allowBroken = true; }
       to ~/.config/nixpkgs/config.nix.
(use '--show-trace' to show detailed location information)
~/D/D/insta_billboard [1]> export NIXPKGS_ALLOW_BROKEN=1
~/D/D/insta_billboard> nix-shell default.nix
these 2 derivations will be built:
  /nix/store/bz6z2h9h0bap02rv1j33yfxvj2q2ksnk-elixir-1.13.4.drv
  /nix/store/qzwi89imacw90ccihrfyr62f4m34c4z2-x11-ssh-askpass-1.2.4.1.drv
these 85 paths will be fetched (86.61 MiB download, 506.77 MiB unpacked):
  /nix/store/06nq0w4f8dg9jvybs57yrr2180hfirdf-zstd-1.5.2-dev
  /nix/store/0abkg7g6crfzjn6n9bv09h6pr7b39rq5-c-ares-1.18.1
  /nix/store/0mzw3caf9jwz1wrsg4f412mmxkrhpn7w-hook
  /nix/store/0n24b06x2brlqgk0s41qm0dd4iysz885-x11-ssh-askpass-1.2.4.1.tar.gz
  /nix/store/15w3ybql6xynyixw3j7pja9pvin99jl3-perl5.36.0-libwww-perl-6.67
  /nix/store/1x4gyddm54853ndv54v6b3rbvk3rp2y3-pkg-config-wrapper-0.29.2
  /nix/store/24hd4g9gs48jxchr5xx61c4g80sy0b63-curl-7.86.0-dev
  /nix/store/2anmjk732z9byhym2dr3crnixcv9rvvb-perl5.36.0-Net-HTTP-6.19
  /nix/store/2cdxfs87d522qan7xswmdwv3rxcmvhm6-perl5.36.0-LWP-MediaTypes-6.04
[... 194 lines cut ...]
ln -s SshAskpass-default.ad SshAskpass.ad
sed -n '/^[^!]/s/.*/"&",/p' <SshAskpass.ad >SshAskpass_ad.h.tmp
mv -f SshAskpass_ad.h.tmp SshAskpass_ad.h
make depend
gccmakedep  --   -I/nix/store/gp8riwg2k6rv8fhmb4i1jg2zaq23q536-xorg-cf-files-1.0.7/include    DarwinMachineDefines -D__DARWIN__                                 -DNO_ALLOCA -DCSRG_BASED         -- drawing.c dynlist.c resources.c x11-ssh-askpass.c
building
build flags: SHELL=/nix/store/x380x9d7gqh4ig9h50a7wx08p0s6622f-bash-5.1-p16/bin/bash
clang -O -Wall -Wpointer-arith -no-cpp-precomp     -I/nix/store/gp8riwg2k6rv8fhmb4i1jg2zaq23q536-xorg-cf-files-1.0.7/include    DarwinMachineDefines -D__DARWIN__                                 -DNO_ALLOCA -DCSRG_BASED          -c -o drawing.o drawing.c
clang-11: error: no such file or directory: 'DarwinMachineDefines'
make: *** [<builtin>: drawing.o] Error 1
error: builder for '/nix/store/qzwi89imacw90ccihrfyr62f4m34c4z2-x11-ssh-askpass-1.2.4.1.drv' failed with exit code 2;
       last 10 log lines:
       > ln -s SshAskpass-default.ad SshAskpass.ad
       > sed -n '/^[^!]/s/.*/"&",/p' <SshAskpass.ad >SshAskpass_ad.h.tmp
       > mv -f SshAskpass_ad.h.tmp SshAskpass_ad.h
       > make depend
       > gccmakedep  --   -I/nix/store/gp8riwg2k6rv8fhmb4i1jg2zaq23q536-xorg-cf-files-1.0.7/include    DarwinMachineDefines -D__DARWIN__                                 -DNO_ALLOCA -DCSRG_BASED         -- drawing.c dynlist.c resources.c x11-ssh-askpass.c
       > building
       > build flags: SHELL=/nix/store/x380x9d7gqh4ig9h50a7wx08p0s6622f-bash-5.1-p16/bin/bash
       > clang -O -Wall -Wpointer-arith -no-cpp-precomp     -I/nix/store/gp8riwg2k6rv8fhmb4i1jg2zaq23q536-xorg-cf-files-1.0.7/include    DarwinMachineDefines -D__DARWIN__                                 -DNO_ALLOCA -DCSRG_BASED          -c -o drawing.o drawing.c
       > clang-11: error: no such file or directory: 'DarwinMachineDefines'
       > make: *** [<builtin>: drawing.o] Error 1
       For full logs, run 'nix log /nix/store/qzwi89imacw90ccihrfyr62f4m34c4z2-x11-ssh-askpass-1.2.4.1.drv'.
error: build of '/nix/store/bz6z2h9h0bap02rv1j33yfxvj2q2ksnk-elixir-1.13.4.drv', '/nix/store/qzwi89imacw90ccihrfyr62f4m34c4z2-x11-ssh-askpass-1.2.4.1.drv' failed
```

fhunleth, 2023-03-14:
Thanks for letting us know about this!

`x11_ssh_askpass` isn't needed on MacOS. I'm not a Nix user, but I'm told that making the `x11_ssh_askpass` conditional like in https://github.com/ceedubs/unison-nix/blob/e00ad743c77174714d1ebbcacf9b8e63cc7f9554/flake.nix#L36 is the way to fix this.

If that makes sense to you and you can test, could you either send a PR or let me know the update you made here so I can fix the docs?

simonelnahas, 2023-03-16:
Would love to test it, although I'm not sure exactly how to incorporate that code snippet you linked into the nix-shell file.

garrettmichaelgeorge, 2023-04-19:
Hi @simonelnahas, are you by chance running macOS on a recent `aarch64` architecture (M1 or M2)? It [looks like](https://github.com/NixOS/nixpkgs/commit/064207b723a0f0e2ccb2fb89a4326e672133905f) `x11_ssh_askpass` was previously marked as broken on `aarch64-darwin`, but the "broken" label was removed in https://github.com/NixOS/nixpkgs/pull/207321 after the package was verified to build on that architecture.

If `x11_ssh_askpass` is still showing as broken (or actually breaking when building locally), then the local version of `nixpkgs` (referenced by `import <nixpkgs>` in the example) probably needs to be updated.

Depending on your preferences for using Nix, a couple ways to update that `nixpkgs` version are:
1. Pin the `nixpkgs` version in `shell.nix` (see the [pinning nixpkgs tutorial](https://nixos.org/guides/towards-reproducibility-pinning-nixpkgs.html) or [this reference](https://nix.dev/reference/pinning-nixpkgs#ref-pinning-nixpkgs)). The following pins it to the most recent commit on `nixpkgs-unstable` as of now:
  ```nix
  { pkgs ? import (fetchTarball "https://github.com/nixos/nixpkgs/archive/0a4206a51b386e5cda731e8ac78d76ad924c7125.tar.gz") {}
  }:
  ```
2. Use a [Nix flake](https://www.tweag.io/blog/2020-05-25-flakes/). Flakes store the nixpkgs version in a lockfile and were invented in part to make this easier.

Let me know if bumping the nixpkgs version solves the issue! If not, then conditionally including the `x11_ssh_askpass` could also work, since it sounds like it is not needed anyway on macOS (no `if`/`then` statement needed):

```nix
{
  # ...
  buildInputs = [
    autoconf
    automake
    curl
    erlangR25
    fwup
    git
    pkgs.beam.packages.erlangR25.elixir
    rebar3
    squashfsTools
    pkg-config
  ] ++ lib.optionals stdenv.isLinux [ x11_ssh_askpass ];
  # ...
}
```