nerves-project/nerves #868: Hardcoded check for GNU Coreutils on MacOS

cloud8421, 2023-05-03:
### Environment

I'm running on MacOS Ventura, using a `flake.nix` file to provision the environment for the project.

Here's the file, which is adapted from the `shell.nix` file provided at https://hexdocs.pm/nerves/installation.html#for-nixos-or-nix-package-manager.

```nix
{
  description = "Robo Clock";

  outputs =
    { self
    , nixpkgs
    , flake-utils
    ,
    }:
    flake-utils.lib.eachDefaultSystem
      (
        system:
        let
          pkgs = import nixpkgs {
            inherit system;
          };

          # Set the Erlang version
          erlangVersion = "erlangR25";
          # Set the Elixir version
          elixirVersion = "elixir_1_14";

          erlang = pkgs.beam.interpreters.${erlangVersion};
          elixir = pkgs.beam.packages.${erlangVersion}.${elixirVersion};
          elixir-ls = pkgs.beam.packages.${erlangVersion}.elixir-ls;
          fwup = pkgs.fwup;
          squashfs = pkgs.squashfsTools;
          coreutils = pkgs.coreutils;
          pkg-config = pkgs.pkg-config;
          autoconf = pkgs.autoconf;
          automake = pkgs.automake;
          curl = pkgs.curl;

        in
        rec {

          devShells.default = pkgs.mkShell {
            buildInputs = [
              erlang
              elixir
              elixir-ls
              fwup
              squashfs
              coreutils
              pkg-config
              autoconf
              automake
              curl
            ];
          };
        }
      );
}
```

### Current behavior

Running `nix develop` produces a valid environment, and I'm able to work on the project. Trying to build a firmware results in this error:

```
** (Mix) gstat (coreutils) is required by the Nerves tooling.
```

I traced this down to https://github.com/nerves-project/nerves/blob/d3d6970028e16b4e46aabb5a0669a69f317c31cd/lib/mix/nerves/preflight.ex#L26, where the preflight check on MacOS looks for `gstat`.

The problem is that on Nix the `coreutils` package doesn't prefix utils with `g` as there's no risk of clashing with system built-ins. In my environment, `stat` is the correct command that would need to be used.

I can work around this by creating a link so that the condition is satisfied, but I am wondering if the preflight check needs to be more sophisticated. If it's ok, we can discuss here and I can provide a PR as a follow-up?

Thanks for the great work!

fhunleth, 2023-05-07:
Hi @cloud8421!

Thanks for bringing this up and sorry for the late reply. I was pretty busy this week.

We have references to `gstat` in a few places. I'd definitely support making `preflight` smarter so that Nerves works better for Nix users. Before you work on that, it would be good to get a sense of the amount of work beyond `preflight`. I mostly remember one place that we use `gstat` a lot when creating the SquashFS. I'm pretty sure there are other places, but I don't recall how wide spread it is. I'll can help get those updates through, but I think it will be a few projects that need changes to properly support Nix.

cloud8421, 2023-05-09:
Thank you - no worries we're all busy!

Following what you said it makes me think of an alternative.

https://github.com/fully-forged/robo_clock/commit/09e4481010b88e4cd212c277a946ad99c2bd829c worked very well for me paired with a temporary extension of `$PATH` to include `bin/gstat`.

I'm thinking if I extend the `flake.nix` file to 1) create this wrapper script 2) place it in `$PATH` that would make things work without having to complicate things at the Nerves level. I would be happy to contribute the Flake file as documentation for setup.

fhunleth, 2023-05-13:
I like that approach a lot! It sounds very simple and is a quick way to help other Nix users running into this. Yes, could you please submit a PR. Thanks!