nerves-project/nerves #893: Failure entering the nerves.system.shell using erlang 26.0.2

samuelventura, 2023-07-15:
### Environment

```
elixir -v
Erlang/OTP 26 [erts-14.0.2] [source] [64-bit] [smp:8:8] [ds:8:8:10] [async-threads:1] [jit:ns]
Elixir 1.14.4 (compiled with Erlang/OTP 26)

cat ~/.tool-versions 
erlang 26.0.2
elixir 1.14.4-otp-26

uname -a
Linux p3420 6.2.0-24-generic #24-Ubuntu SMP PREEMPT_DYNAMIC Fri Jun 16 12:03:50 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

### Current behavior

The task `nerves.system.shell` fails on a customized x64 nerves system upgraded to  release [v1.23.0](https://github.com/nerves-project/nerves_system_x86_64/releases/tag/v1.23.0).

```
$ mix nerves.system.shell
sh: 1: exec: tty_sl: not found
** (ArgumentError) argument error
    :erlang.port_control(#Port<0.8>, 25889124, [])
    (nerves 1.10.2) lib/mix/nerves/shell.ex:106: Mix.Nerves.Shell.get_tty_geometry/1
    (nerves 1.10.2) lib/mix/nerves/shell.ex:49: Mix.Nerves.Shell.open/2
    (nerves_bootstrap 1.11.5) lib/attic/system.shell.ex:74: Mix.Tasks.Nerves.System.Shell.run/1
    (mix 1.14.4) lib/mix/task.ex:421: anonymous fn/3 in Mix.Task.run_task/4
    (mix 1.14.4) lib/mix/cli.ex:84: Mix.CLI.run_task/2
```

### Expected behavior

The nerves shell to customize a system should show up.

### Other findings

According to the links below, the erlang release 26.0.2 dropped [erts/emulator/drivers/unix/ttsl_drv.c](https://github.com/erlang/otp/commit/bd0865ff86f06bba13a99e38db5d1f0a124e5cb3#diff-0576e6bef71cef2f053adbdba2a0a4cee12d617e04a47da84617988fd5cf829b) which compiles to `tty_sl` need by [lib/mix/nerves/shell.ex](https://github.com/nerves-project/nerves/blob/570ab2a816eb8ad76a35a8227e20501d431248a9/lib/mix/nerves/shell.ex#L17).

- https://github.com/erlang/otp/releases/tag/OTP-26.0.2
- https://github.com/erlang/otp/commit/d051172925a5c84b2f21850a188a533f885f201c
- https://github.com/erlang/otp/tree/d051172925a5c84b2f21850a188a533f885f201c/erts/emulator/drivers/unix

fhunleth, 2023-07-15:
Hi @samuelventura! Thank you for the report. We're working our way to fixing this issue. There have been several cases where tty updates broke libraries and @jjcarstens and others have been fixing them. 

I'm not sure when we'll have this fixed. In the mean time, if you have a Linux machine available, you can build Nerves systems using https://github.com/nerves-project/nerves_systems.

jjcarstens, 2023-09-13:
Temporary workaround in https://github.com/nerves-project/nerves/pull/897