nerves-project/nerves #927: Running `mix compile.nerves_package` crashes

Noarkhh, 2023-11-06:
### Environment

* Elixir version: 1.15.6
* Nerves environment: rpi4, dev
* Host OS: MacOS and Ubuntu

### Current behavior

After running `mix compile.nerves_package` in a new project the compilation crashes:
```
mix nerves.new test
cd test
export MIX_TARGET=rpi4
mix deps.get
mix compile.nerves_package
```
stacktrace:
```
** (KeyError) key :type not found in: nil

If you are using the dot syntax, such as map.field, make sure the left-hand side of the dot is a map
    (nerves 1.10.4) lib/nerves/artifact.ex:210: Nerves.Artifact.env_var/1
    (nerves 1.10.4) lib/nerves/artifact.ex:200: Nerves.Artifact.env_var?/1
    (nerves 1.10.4) lib/nerves/artifact.ex:89: Nerves.Artifact.stale?/1
    (nerves 1.10.4) lib/mix/tasks/compile.nerves_package.ex:28: Mix.Tasks.Compile.NervesPackage.run/1
    (mix 1.15.6) lib/mix/task.ex:447: anonymous fn/3 in Mix.Task.run_task/5
    (mix 1.15.6) lib/mix/cli.ex:92: Mix.CLI.run_task/2
```

fhunleth, 2023-11-06:
Could you run `mix firmware`?

Out of curiosity, where are you seeing the reference to `mix compile.nerves_package`? We need to fix this, but it's not the normal way to use Nerves, so I'd like to update that reference. Thank you for reporting it!

closed 2023-11-08 by jjcarstens

Noarkhh, 2023-11-11:
Hi, thanks for quick response!

I found it [here](https://hexdocs.pm/nerves/Mix.Tasks.Compile.NervesPackage.html), in the official documentation. I guess some broader documentation could be useful. I was messing around with tasks when trying to locate binaries not contained within the artifact located in `~/.nerves/artifacts/<my_artifact>/`, but present in the generated firmware. Would it be possible for you to lend me a hand in this regard? If so you can suggest an appropriate place to continue our conversation.

fhunleth, 2023-11-11:
Got it. Thanks. I refreshed the hexdocs to include @jjcarstens's documentation fix. 

The best places to go for help are to either post something on the Elixir forum (add the Nerves label for someone to see it)  or join the Elixir Discord or Slack and post to the #nerves channel.

PR #928 (2023-11-08): Ensure a package is loaded before `compile.nerves_package`
Fixes #927

This previously ignored the return of `Nerves.Env.ensure_loaded/1` which could actually return an `{:error, _}` tuple. When calling the `compile.nerves_package` manually, it would crash with a confusing error.

This instead checks the return and raises on error.

It also adjusts the documentation on how/when this task is used
[image]
commit 8863548c (2023-11-08): Ensure a package is loaded before `compile.nerves_package`

Fixes #927

This previously ignored the return of `Nerves.Env.ensure_loaded/1` which
could actually return an `{:error, _}` tuple. When calling the
`compile.nerves_package` manually, it would crash with a confusing error.

This instead checks the return and raises on error.

It also adjusts the documentation on how/when this task is used