nerves-project/nerves #937: Discourage mktemp usage

salseeg, 2023-12-16:
### Environment

* Elixir version (`elixir -v`): 
```
Erlang/OTP 26 [erts-14.1] [source] [64-bit] [smp:12:12] [ds:12:12:10] [async-threads:1] [jit:ns] 

Elixir 1.15.5 (compiled with Erlang/OTP 26)
```
* Nerves environment: (`mix nerves.env --info`)
```
nerves| Environment Package List

  Pkg:         nerves_toolchain_ctng
  Vsn:         1.9.3
  Type:        toolchain_platform
  BuildRunner: {nil, []}

  Pkg:         nerves_toolchain_aarch64_nerves_linux_gnu
  Vsn:         1.8.0
  Type:        toolchain
  BuildRunner: {Nerves.Artifact.BuildRunners.Local, []}

  Pkg:         bktp_rpi4
  Vsn:         1.24.1-bktp.3
  Type:        system
  BuildRunner: {Nerves.Artifact.BuildRunners.Docker, [make_args: ["source", "all", "legal-info"]]}

  Pkg:         nerves_system_br
  Vsn:         1.24.1
  Type:        system_platform
  BuildRunner: {nil, []}

|nerves| Loadpaths Start

Nerves environment
  MIX_TARGET:   bktp_rpi4
  MIX_ENV:      prod

|nerves| Environment Variable List
  target:     bktp_rpi4
  toolchain:  /Users/s/.nerves/artifacts/nerves_toolchain_aarch64_nerves_linux_gnu-darwin_x86_64-1.8.0
  system:     /Users/s/.nerves/artifacts/bktp_rpi4-portable-1.24.1-bktp.3
  app:        .

|nerves| Loadpaths End

Nerves:           1.10.4
Nerves Bootstrap: 1.12.1
Elixir:           1.15.5
|nerves| Info End
```
* Additional information about your host, target hardware or environment that
  may help: `/Volume/cs_fs` is case sensitive on my Mac. `/` (and `/tmp` cosequently) is case insensitive.

### Current behavior

Original error on `mix firmware`
```
|nerves| Building OTP Release...

* [Nerves] validating vm.args
* skipping runtime configuration (config/runtime.exs not found)
* creating _build/bktp_rpi4_prod/rel/platform/releases/2023-12-14_10469b5___2023-12-14_3215ba1/vm.args
Updating base firmware image with Erlang release...
Copying rootfs_overlay: /Volumes/cs_fs/buckitUp/platform/_build/bktp_rpi4_prod/nerves/rootfs_overlay
Copying rootfs_overlay: /Volumes/cs_fs/buckitUp/platform/rootfs_overlay

FATAL ERROR: write_file: file squashfs-root/usr/lib/xtables/libip6t_hl.so already exists
```

`/Volume/cs_fs` is case sensitive on my Mac. `/` (and `/tmp` cosequently) is case insensitive.

I faces problems building any custom system before. Workaround of moving linux kernel to  case sensitive FS worked.

This time I was adding `iptables` and described error have happend.
`iptables` generates `` and  ``. This is the root cause of the error.
```bash
ls usr/lib/xtables | grep ip6t_

libip6t_DNPT.so
libip6t_HL.so
libip6t_NETMAP.so
libip6t_REJECT.so
libip6t_SNPT.so
libip6t_ah.so
libip6t_dst.so
libip6t_eui64.so
libip6t_frag.so
libip6t_hbh.so
libip6t_hl.so
libip6t_icmp6.so
libip6t_ipv6header.so
libip6t_mh.so
libip6t_rt.so
libip6t_srh.so
```

Moving project to case sensitive FS had not helped.

See Workaround down below.

### Expected behavior

Create custom temp directory under project `_build` or Nerves dir. This way developer may control the filesystem used for building.

It would be helpful to note in the documentation that case sensitive filesystem is needed to build.

### Workaround

Replacing `mktemp` usage with hardcoded directory in `merge-squashfs` worked.
```bash
...
workdir="/Volumes/cs_fs/nerves/squash_merge_tmp"
rm -rf "$workdir"
mkdir -p "$workdir"
# workdir=$(mktemp -d 2>/dev/null || mktemp -d -t 'merge-squashfs-tmp')
pushd "$workdir" >/dev/null
...
```

salseeg, 2023-12-16:
Added as instruction in my project - https://github.com/Buckitup-chat/bktp_rpi4/tree/otp_26?tab=readme-ov-file#macos-specific

fhunleth, 2024-08-20:
@salseeg Sorry for the delayed response and thank you for sharing your instructions. We've been very slowly working on a way to remove the use of `merge-squashfs`, and I had hoped to share it by now, but that hasn't happened. The issue with adding `iptables` and building on MacOS has hit quite a few people.