nerves-project/nerves #942: Load Nerves.Runtime.KV in runtime.exs crashes system

guillego, 2023-12-29:
### Environment

* Elixir version (`elixir -v`):
Erlang/OTP 26 [erts-14.0.2] [source] [64-bit] [smp:8:8] [ds:8:8:10] [async-threads:1] [jit]
Elixir 1.14.5 (compiled with Erlang/OTP 26)

* Nerves environment:
  MIX_TARGET:   rpi4
  MIX_ENV:  dev

### Current behavior
I'm trying to load some config values into runtime.exs from the unique device u-boot key-value metadata like this:

runtime.exs
```elixir

import Config

config :system_info, :device,
  name: Nerves.Runtime.KV.get("device_name"),
  id: Nerves.Runtime.KV.get("clutchcam_id")
```

If I don't put this in runtime.exs, the device starts normally, and I can run the KV commands through iex  without issues.

However if I put that in the runtime.exs the device is completely unresponsive. By burning an upgrade without the runtime file I was able to log in again and recover the erl_crash.dump which contains this.

```erlang
Slogan: Runtime terminating during boot ({noproc,{'Elixir.GenServer',call,['Elixir.Nerves.Runtime.KV',{get,<<"clutchcam_name">>},5000]}})
```

Is the KV not available when the runtime config is loaded? Is there a way to enable this or to load the KV values into the config through some other mechanism?

jjcarstens, 2023-12-29:
[`runtime.exs` is evaluated after core Elixir/Erlang applications have started but before any of your applications have started](https://hexdocs.pm/mix/Mix.Tasks.Release.html#module-runtime-configuration). That means if you want `Nerves.Runtime.KV` there then you will need to manually call `Application.ensure_all_started(:nerves_runtime)` before making any KV calls.

That said, if you don't have a revert mechanism then I would still recommend doing this elsewhere (such as your `application.ex`) in order to handle any error more gracefully if needed

closed 2023-12-29 by jjcarstens

fhunleth, 2023-12-29:
It seems possible to change `Nerves.Runtime.KV` to  be callable before the application is started. If I recall, the issue here is that values are cached in a GenServer. If the caching is removed, I'd expect this to work. In hindsight, I suspect that caching KV values isn't much of a win anyway, but I'd want to double check. If this is important to you, please feel free to start a PR.

guillego, 2023-12-29:
Thanks @jjcarstens and @fhunleth! It's not critical to load the config in the runtime.exs, but it seemed like the most elegant solution for my case. But it's interesting to know that it could potentially be removed, I'll dig through `nerves_runtime` and see what I find.

I'll also take a look at implementing this in application.ex, it makes sense to have a better error handling that doesn't leave the system unusable.