nerves-project/nerves #977: `nerves.system.shell` crashes on macos

Noarkhh, 2024-05-23:
### Environment

* Elixir version: 1.16
* MacOS 14.4, M3, 

### Current behavior

When running `nerves.system.shell` inside `nerves_system_rpi4` I get the following error:
```
|Elixir.Docker| Starting Build... (this may take a while)

** (Mix) The Nerves Docker build_runner encountered an error while building:

-----
docker: Error response from daemon: error while creating mount source path '/host_mnt/private/tmp/com.apple.launchd.1KMZvn8mWg/Listeners': mkdir /host_mnt/private/tmp/com.apple.launchd.1KMZvn8mWg/Listeners: operation not supported.

-----

See /Users/noarkhh/Nerves/nerves_system_rpi4_ex_webrtc/build.log.
```

This has to be some problem with ssh forwarding to docker on macos, since when I removed `ssh_agent()` from [this_line](https://github.com/nerves-project/nerves/blob/5eb695365786f30a843e7902c6e087325f896654/lib/nerves/artifact/build_runners/docker.ex#L247) things started working.

I think this might have something to do woth this issue https://github.com/docker/for-mac/issues/410

fhunleth, 2024-05-23:
Hi @Noarkhh - Could you try out #972 and see if it works for you?

Noarkhh, 2024-06-04:
Hi @fhunleth, seems to work, thanks for the quick response!

fhunleth, 2024-06-06:
@Noarkhh Thanks for following up on this! Always good to know that things work.

closed 2024-06-06 by fhunleth