nerves-project/nerves #1001: `No space left on device` when `mix nerves.system.shell`

xmyst, 2024-08-08:
### Environment

* Elixir version (`elixir -v`):
  ```
  Erlang/OTP 27 [erts-15.0.1] [source] [64-bit] [smp:8:8] [ds:8:8:10] [async-threads:1]

  Elixir 1.17.2 (compiled with Erlang/OTP 27)
  ```
* Nerves environment (`mix nerves.env --info`):
  ```
  % mix nerves.env --info

  ==> nerves
  ==> custom_rpi4
  Generated custom_rpi4 app
  |Elixir.Docker| Starting Build... (this may take a while)

  ..** (Mix) The Nerves Docker build_runner encountered an error while building:

  -----
  rpicam-apps (BR2_PACKAGE_RPICAM_APPS) [N/y/?] (NEW) 
  rpi-distro-bluez-firmware (BR2_PACKAGE_RPI_DISTRO_BLUEZ_FIRMWARE) [N/y/?] (NEW) 
  rpi-distro-firmware-nonfree (BR2_PACKAGE_RPI_DISTRO_FIRMWARE_NONFREE) [N/y/?] (NEW) 
  *
  * rpi-libcamera
  *
  rpi-libcamera (BR2_PACKAGE_RPI_LIBCAMERA) [N/y/?] (NEW) 
  #
  # configuration written to /home/nerves/project/.config
  #

  *** Error during update of the configuration.

  make[2]: *** [Makefile:1036: syncconfig] Error 1
  make[1]: *** [Makefile:601: /home/nerves/project/build/buildroot-config/auto.conf] Error 2
  make: *** [Makefile:23: _all] Error 2
  -----
  ```
* Additional information about your host, target hardware or environment that may help:
  - Nerves dependency is `{:nerves, github: "nerves-project/nerves", runtime: false, override: true}`.

### Current behaviour

```
% mix nerves.system.shell
==> nerves
==> pcblabs_rpi4
Generated pcblabs_rpi4 app
|Elixir.Docker| Starting Build... (this may take a while)

.
>>> host-tar 1.35 Extracting** (Mix) The Nerves Docker build_runner encountered an error while building:

-----
cpio: tar-1.35/tests/extrac24.at not created: newer or same age version exists
cpio: tar-1.35/tests/sptrdiff00.at not created: newer or same age version exists
cpio: tar-1.35/tests/T-nonl.at not created: newer or same age version exists
cpio: tar-1.35/tests/listed04.at not created: newer or same age version exists
cpio: tar-1.35/tests/rename03.at not created: newer or same age version exists
cpio: tar-1.35/tests/ignfail.at not created: newer or same age version exists
cpio: tar-1.35/tests/link01.at not created: newer or same age version exists
cpio: tar-1.35/tests/exclude07.at not created: newer or same age version exists
cpio: tar-1.35/tests/remfiles09c.at not created: newer or same age version exists
cpio: tar-1.35/tests/extrac18.at not created: newer or same age version exists
cpio: tar-1.35/tests/sparse01.at not created: newer or same age version exists
cpio: tar-1.35/tests/multiv04.at not created: newer or same age version exists
cpio: tar-1.35/tests/testsuite not created: newer or same age version exists
cpio: write error: No space left on device
make[1]: *** [package/pkg-generic.mk:207: /home/nerves/project/build/host-tar-1.35/.stamp_extracted] Error 2
make: *** [Makefile:23: _all] Error 2
-----
```

### Expected behaviour

OTP 26 warning and a long Docker command.

jjcarstens, 2024-08-08:
How much space are you allocating for docker images? This is dependent on your settings as it mounts volumes up to the maximum of the docker limit. It could be as simple as increasing the amount of space available there.

It also could be your actual drive of this computer is at it's limit as well

xmyst, 2024-08-08:
### Docker

RAM: 8 GB
Swap: 1 GB
Virtual disk limit: 64 GB

I am going to bump the disk to 128 GB and report back.

### Local disk

```
% df -ahY

Filesystem      Type      Size    Used   Avail Capacity iused ifree %iused  Mounted on
/dev/disk1s5s1  apfs     233Gi   9.6Gi    88Gi    10%    404k  921M    0%   /
devfs           devfs    190Ki   190Ki     0Bi   100%     656     0  100%   /dev
/dev/disk1s2    apfs     233Gi   2.0Gi    88Gi     3%    6.7k  921M    0%   /System/Volumes/Preboot
/dev/disk1s4    apfs     233Gi   1.0Gi    88Gi     2%       1  921M    0%   /System/Volumes/VM
/dev/disk1s6    apfs     233Gi   3.9Mi    88Gi     1%      25  921M    0%   /System/Volumes/Update
/dev/disk1s1    apfs     233Gi   132Gi    88Gi    60%    1.1M  921M    0%   /System/Volumes/Data
map auto_home   autofs     0Bi     0Bi     0Bi   100%       0     0     -   /System/Volumes/Data/home
```

xmyst, 2024-08-08:
Finished successfully with 128 GB disk limit. Apparently, I have been walking on thin ice all this time 🤷🏻‍♂️

closed 2024-08-08 by xmyst