nerves-project/nerves #1004: Nerves Shell Broken

CaptChrisD, 2024-08-30:
### Environment

Elixir 17.0.2
Erlang 27.0.1
macOS 14.2.1

### Current behavior

The Nerves Shell Docker container runs but I am unable to configure the system with `make menuconfig` or `make linux-menuconfig`

`make menuconfig` seems to partially load the Buildroot configuration tool but I only see a line of text along the bottom
`make linux-menuconfig` seems to start building linux rather then load the configuration tool

### Expected behavior

Should have the same behavior as before where the system can be configured

fhunleth, 2024-08-30:
@CaptChrisD I transferred the issue to Nerves since it seems like it should be a generic issue. Do you have any screenshots or logs or anything else you can share?

dylanfareed, 2024-08-31:
Hi @fhunleth. I believe I'm running into the same or a similar issue so I'll try to provide some context.

Locally I'm running 1.17.2/27 on a mac.

```
> iex -v
Erlang/OTP 27 [erts-15.0.1] [source] [64-bit] [smp:10:10] [ds:10:10:10] [async-threads:1] [jit]

IEx 1.17.2 (compiled with Erlang/OTP 27)
```

Here's the docker details

```
> docker -v
Docker version 27.1.1, build 6312585
```

I'm creating a custom system by cloning the rpi4 system

```
> git clone https://github.com/nerves-project/nerves_system_rpi4.git custom -b v1.28.1
```

Running `mix nerves.system.shell` returns fine with the expected (iirc) docker command.

[image]

Nerves Shell boots fine seemingly

[image]

Then when trying to run either `make menuconfig` or `make linux-menuconfig` they start to build and then just hang seemingly forever.

[image]

Here's a screenrecording
https://www.loom.com/share/682da9ad95894ca3bc83091f0fc02ee4

Here's the outputs for each

```
 Nerves  project > make menuconfig
  GEN     /home/nerves/project/Makefile

/home/nerves/project/.config - Buildroot 2024.05.2 Configuration────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
```

```
 Nerves  project > make linux-menuconfig
/usr/bin/make -j1  O=/home/nerves/project HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
  GEN     /home/nerves/project/Makefile
GIT_DIR=. BR_BINARIES_DIR=/home/nerves/project/images KBUILD_BUILD_VERSION=1 KBUILD_BUILD_USER=buildroot KBUILD_BUILD_HOST=buildroot KBUILD_BUILD_TIMESTAMP="Wed Aug 14 08:43:20 UTC 2024" KCFLAGS=-Wno-attribute-alias DTC_FLAGS=-@ PKG_CONFIG_PATH="" /usr/bin/make -j11 -C /home/nerves/project/build/linux-custom HOSTCC="/usr/bin/gcc" HOSTCC="/usr/bin/gcc -O2 -isystem /home/nerves/project/host/include -L/home/nerves/project/host/lib -Wl,-rpath,/home/nerves/project/host/lib" ARCH=arm64 INSTALL_MOD_PATH=/home/nerves/project/target CROSS_COMPILE="/home/nerves/project/host/bin/aarch64-nerves-linux-gnu-" WERROR=0 REGENERATE_PARSERS=1 DEPMOD=/home/nerves/project/host/sbin/depmod INSTALL_MOD_STRIP=1 HOSTCC="/usr/bin/gcc" menuconfig

.config - Linux/arm64 6.6.31 Kernel Configuration-----------------------------------------------------------------------------------------------------------------------------------------------
```

Hopefully that's helpful. Please lmk if there's any other details I can provide. Thanks!

dylanfareed, 2024-09-01:
Just want to confirm that the same thing happens for `make busybox-menuconfig`

Running it within the shell appears to build successfully and then nothing ever launches.

[image]

```
 Nerves  project > make busybox-menuconfig
/usr/bin/make -j1  O=/home/nerves/project HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
  GEN     /home/nerves/project/Makefile
busybox-1.36.1.tar.bz2: OK (sha256: b8cc24c9574d809e7279c3be349795c5d5ceb6fdf19ca709f80cde50e47de314)
>>> host-skeleton  Extracting
>>> host-skeleton  Patching
>>> host-skeleton  Configuring
>>> host-skeleton  Building
>>> host-skeleton  Installing to host directory
# check-package DoNotInstallToHostdirUsr
>>> host-fakedate  Extracting
>>> host-fakedate  Patching
>>> host-fakedate  Configuring
>>> host-fakedate  Building
>>> host-fakedate  Installing to host directory
/usr/bin/install -D -m 755 package/fakedate/fakedate /home/nerves/project/host/bin/date
tar-1.35.cpio.gz: OK (sha256: c77a38fcf25b21fd8209d20d35638744344ded239cfc7df80138bf46d3c6b16d)
>>> host-tar 1.35 Extracting
mkdir -p /home/nerves/project/build/host-tar-1.35
cd /home/nerves/project/build/host-tar-1.35 && gzip -d -c /nerves/dl/tar/tar-1.35.cpio.gz | cpio -i --preserve-modification-time
39210 blocks
mv /home/nerves/project/build/host-tar-1.35/tar-1.35/* /home/nerves/project/build/host-tar-1.35
rmdir /home/nerves/project/build/host-tar-1.35/tar-1.35
>>> host-tar 1.35 Patching

Applying 0001-lib-getrandom.c-fix-build-with-uclibc-1.0.35.patch using patch: 
patching file gnu/getrandom.c
>>> host-tar 1.35 Updating config.sub and config.guess
for file in config.guess config.sub; do for i in $(find /home/nerves/project/build/host-tar-1.35 -name $file); do cp support/gnuconfig/$file $i; done; done
>>> host-tar 1.35 Patching libtool
>>> host-tar 1.35 Configuring
(cd /home/nerves/project/build/host-tar-1.35/ && rm -rf config.cache; GIT_DIR=. PATH="/home/nerves/project/host/bin:/home/nerves/project/host/sbin:/home/nerves/bin:/home/nerves/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" PKG_CONFIG="/home/nerves/project/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/home/nerves/project/host/lib/pkgconfig:/home/nerves/project/host/share/pkgconfig" AR="/usr/bin/ar" AS="/usr/bin/as" LD="/usr/bin/ld" NM="/usr/bin/nm" CC="/usr/bin/gcc" GCC="/usr/bin/gcc" CXX="/usr/bin/g++" CPP="/usr/bin/cpp" OBJCOPY="/usr/bin/objcopy" RANLIB="/usr/bin/ranlib" CPPFLAGS="-I/home/nerves/project/host/include" CFLAGS="-O2 -I/home/nerves/project/host/include" CXXFLAGS="-O2 -I/home/nerves/project/host/include" LDFLAGS="-L/home/nerves/project/host/lib -Wl,-rpath,/home/nerves/project/host/lib" INTLTOOL_PERL=/usr/bin/perl CFLAGS="-O2 -I/home/nerves/project/host/include" LDFLAGS="-L/home/nerves/project/host/lib -Wl,-rpath,/home/nerves/project/host/lib" CC="/usr/bin/gcc" CXX="/usr/bin/g++" CONFIG_SITE=/dev/null ./configure --prefix="/home/nerves/project/host" --sysconfdir="/home/nerves/project/host/etc" --localstatedir="/home/nerves/project/host/var" --enable-shared --disable-static --disable-gtk-doc --disable-gtk-doc-html --disable-doc --disable-docs --disable-documentation --disable-debug --with-xmlto=no --with-fop=no --disable-nls --disable-dependency-tracking  --without-selinux )
configure: WARNING: unrecognized options: --enable-shared, --disable-static, --disable-gtk-doc, --disable-gtk-doc-html, --disable-doc, --disable-docs, --disable-documentation, --disable-debug, --with-xmlto, --with-fop
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether UID '1005' is supported by ustar format... yes
checking whether GID '1005' is supported by ustar format... yes
checking how to create a ustar tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking for gcc... /usr/bin/gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether /usr/bin/gcc accepts -g... yes
checking for /usr/bin/gcc option to enable C11 features... none needed
checking whether /usr/bin/gcc understands -c and -o together... yes
checking whether the compiler is clang... no
checking for compiler option needed when checking for declarations... none
checking whether make supports the include directive... yes (GNU style)
checking dependency style of /usr/bin/gcc... none
checking for ranlib... /usr/bin/ranlib
checking for bison... no
checking for byacc... no
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
[... 2567 lines cut ...]
klogd (5.7 kb) (KLOGD) [N/y/?] n
logger (6.3 kb) (LOGGER) [N/y/?] n
logread (4.8 kb) (LOGREAD) [N/y/?] n
syslogd (13 kb) (SYSLOGD) [N/y/?] n
/home/nerves/project/build/busybox-1.36.1/scripts/gcc-version.sh: line 11: /home/nerves/project/host/bin/aarch64-nerves-linux-gnu-gcc: No such file or directory
[... cut ...]
  HOSTCC  scripts/kconfig/lxdialog/menubox.o
  HOSTCC  scripts/kconfig/lxdialog/msgbox.o
  HOSTCC  scripts/kconfig/lxdialog/textbox.o
  HOSTCC  scripts/kconfig/lxdialog/util.o
  HOSTCC  scripts/kconfig/lxdialog/yesno.o
  HOSTLD  scripts/kconfig/lxdialog/lxdialog
scripts/kconfig/mconf Config.in
#
# using defaults found in .config
#

BusyBox 1.36.1 Configuration--------------------------------------------------------------------------------
```

If there's a specific handy log to check out, I'm happy to debug further. Thanks!

fhunleth, 2024-09-01:
Thank you for so much context!!! Let me study. In case it's not obvious, I do all of my Nerves system work on Linux. On the outside chance that anyone reading this loves Docker and wants to help keep everything working well in MacOS, please don't be bashful.

I'm in and out this weekend, so I'll probably get to posting back Tuesday unless there's something really obvious in the logs.

fhunleth, 2024-09-01:
Here's a quick fix. Run the following at the command prompt before running `make menuconfig`:

```sh
export TERM=xterm-256color
```

Next step for me is to figure out where best to put this or what happened.

dylanfareed, 2024-09-01:
Thanks @fhunleth. Just wanted to confirm that that fixes it locally for me as well. 🚀

fhunleth, 2024-09-01:
Awesome! Thanks for confirming.

CaptChrisD, 2024-09-02:
@fhunleth Yes, that fixes the issues on my system as well. Thanks for the prompt response!

joevandyk, 2024-09-25:
The workaround of setting `export TERM=xterm-256color` also worked for me on macos.

commit 65d3741c (2024-09-26): Force `TERM` for docker system shell

Fixes #1004

This explicitly sets the `TERM` var for system shells using docker so
that the config menus will correctly render
PR #1011 (2024-09-27): Force `TERM` for docker system shell
Fixes #1004

EDIT:
Turns out we [added `TERM=dumb` to the `nerves_system_br` image starting with v1.28](https://github.com/nerves-project/nerves_system_br/commit/db6f9a2a5848ca75f26c9c4a3b2a7bf80e606da1#diff-181ba968e8b37dc7c64319846e544967221e3b5c1ec96bd070456abe37b2ccf2R17) based on CircleCI Docker template recommendations. This is nice to have for CI envs which is the majority of uses for that image, so this adjusts the Nerves tooling to put the `TERM` value of the user in _or_ default to `xterm-256color` if not set
commit f5a80410 (2024-09-27): Force `TERM` for docker system shell

Fixes #1004

This explicitly sets the `TERM` var for system shells using docker so
that the config menus will correctly render