nerves-project/nerves #1041: Support Nerdctl/Podman

ananthb, 2025-01-07:
### Environment

MacOS running on Apple Silicon

* Elixir version (`elixir -v`): 1.17.3
* Nerves environment: (`mix nerves.env --info`) 
* Additional information about your host, target hardware or environment that
  may help

### Current behavior

Include errors, stacktraces, screenshots or code that may help us reproduce the
issue.

When I run `mix nerves.system.shell` I get this error message.

```
** (Mix) Docker is not installed on your machine.
Please install docker ~> 1.12 or ~> 1.12.0-rc2 or >= 17.0.0 or later
```

I don't have Docker installed so that makes sense.

### Expected behavior

I do have nerdctl installed which is a drop-in, fully compatible replacement for Docker. I'd like to contribute support for that and additionally for building nerves systems with Podman as well.

ananthb, 2025-01-07:
I need a bit of help locating the code which actually uses Docker to run the builds.

jjcarstens, 2025-01-15:
@ananthb You have a couple routes:
* if `nerdctl` is a full drop-in, then you could simply make the executable configurable (or automatically detected) when shelling out in `Nerves.Artifact.BuildRunners.Docker` (i.e. adjust lines [like this](https://github.com/nerves-project/nerves/blob/main/lib/nerves/artifact/build_runners/docker.ex#L155)).
* `Nerves.Artifact.BuildRunner` is the behavior which can be implemented for different wants to build. That could be a copy/paste of `Nerves.Artifact.BuildRunners.Docker`

That said, for the drop-in replacements I think the first option is probably simplest, but either works

ananthb, 2025-01-16:
Yeah that's perfect. I'll tackle enabling the drop-in replacements first.

Once I'm more comfortable with Elixir and Nerves I want to switch this over to use the OCI API.
That should unlock even wider compatibility.